Moddable-OpenSource/moddable #642: print(5n ** 5n ** 6n) crashes xst (in GC?)

dckc, 2021-05-15:
**Steps to Reproduce**

```
moddable/xs/makefiles/lin$ xst -e 'print(5n ** 5n ** 6n)'
Segmentation fault (core dumped)
```

stack trace:
```
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:435
#1  0x000056356a2f09d4 in fxSweep ()
#2  0x000056356a2f1555 in fxNewChunk ()
#3  0x000056356a282803 in fxBigInt_ulsl1 ()
#4  0x000056356a28781f in fxBigInt_udiv ()
#5  0x000056356a288468 in fxBigintToString ()
#6  0x000056356a265f2e in fxToString ()
#7  0x000056356a35de28 in fx_print ()
#8  0x000056356a30b1a0 in fxRunID ()
#9  0x000056356a318cf3 in fxRunScript ()
#10 0x000056356a35d328 in fx_evalScript ()
#11 0x000056356a30b1a0 in fxRunID ()
#12 0x000056356a2628c9 in main ()
```

**Build environment:** Linux
**Target device:** xst

2674e932f86564d0476f03316eaa744a8edf464d

related: https://github.com/Agoric/agoric-sdk/issues/2652

dckc, 2021-05-15:
@phoddie I don't know how urgent this is for Agoric, but it seems to be a straightforward correctness issue, so it's clearly important. I would appreciate knowing whether it looks like an easy fix or not.

cc @warner @kriskowal @dtribble

phoddie, 2021-05-15:
Curious. It runs without an exception in the Moddable SDK helloworld example. It prints a really big number:

```
2548694315362600004147850217806039692028438907780733491000143581203716887207823195712053711123167319306805836342754527893226911374272580684313221075236201239985451509728931010314858173000857803721033968798295643047379175856372319261603002230628399654515696297591726006820661424424380650449284195785520152690238164298449120852410629500883395876421396246549987659965474197037166981873189906515917468418393104252276206411383581070900414024794869297436499066154525576014416796340846318890953154915136853336174158439809692594628986261972664060392421940125673383370223607001404816345662233843867064158359397820848040961195456700957032095669535408765728965281630257963319105926788682786699907192904731277579081347584188656288944576507507372220896615093559706789620218228301056415364227100477909786431820878308683834360779442519106602063194769589975081584160028861633305859603196928235861402757020354793944023878848890663733356217635850699177056842622612600856086989866145971626530540543511400926148007154657409298243575421273040344248638502514794408896715640830453991799202219305245739046289322078593186450258896193152690687485113873294983456879257908226783834555593159938221728358747971766162788839125738808589686325494660371399688583737049277450720498213689256170243973114789314512151835753487292679921373325421531970352712194998972858108940865473220719996319570447579236568913101630114283835565835416629582792744293020895041266516301610108705188345255673686846405588309397730494091921967454131089800259387184154923364914872169735993256962529925615153847509288941842497262032650204956346807872197230715747113415620684791424315227976523980303938387523241833607661555816564868083138924929918936911767765608671300645858717851394001872051794111496488627983162577743587084663517198982935784785200610912381247204729733412379380611738594445597137928387476240304392743492041618473812528533347883254013555259759947334404610024795501686326473159184704898522905785042619807827790558455698594923332096243813824493238880635725871556031991609584682812802820339502227202396299661255956986126838010174469980141921740634390955487097926147079549817771241285725465982725053057353206243095232588930966081483334873613358386837425131865713891436357345323639997786057709477184789688359052618354432283520901556529209641036944295131667019136080741031143318896816037319283317002161748021362129818940439203393093271874502572820943530559826938787993589916361862467030802538777414966362604230979460291611284704253984499263699227132251388559364676471483009752524004385583964034345496657258674702248593592361756886558994172705714248143352909861556010142823658881609365203020680645508867159698880875061338340436202566450713065659231205488602374782781332861103146434003061181991706250149098077057902760454770011208184851385753865564147895328093701654049017889090530446031884767620582281078735691017290692967396334476751984080802873378828083845307565729142572988962909588943457109345591141947990092080891823013186926926921931059871801796189812890672022029612865677371752757272160418925913760858611859019870451619053956664803738300841882803897191587391500042726905877136817924790131367239419508105025594679944747204574987324372637631504207760959757392403364856551673250689125290450849216806405676845290700285568324201987922643549394663032991794419099351231256598640970113891955781274871854274072425501051377928607481763155296869631490233787049415300265873321501862268247043626062296361979274022756913066016074543837227094603392567829913989290018966145031558038875597572447977506376450734367194972696162083391655186883244630756915020474821181013779903168243682450607005727514663057666755515063397219099918133547339636092728209912918797944426751398096417885050596338429591890469857553263022444613878440132935483118309656199077114586444977861415508896519697277709528452075184147752136299333148857825381063990243163090116157160982873493183237574871117997251931799265200845975955077053797785924410100921094715195989222458805307249473418585382330602590158234887644207210705591492755734522109860923025371370052403352324648720497788071835522779648406714468163610360038122920786114735227420861643457053008588771424667590093588159940553196400037329182834266631371070878557795670247465652718524977449230800410724854608468162439870065125877082886672124461342900905855333111489207145059590409903048053109428531527782563414952937352045746948851156918283493158934790903351824486665027142012212056249612513661613875750805055746999041663582140431462128361733918098808463250880483398076743508341649285645004797276450803029366908327112797557265973224705677478393211457149532264529707609835712330866855661844514463615585944303204589475153717151816381568624492589469462749773421572183070790117771310753475159967598982217150700615460165917167251354657802902698236856481631303597967548338118762724923947552891440436577501242179305595863605601626257069643304567016286206671321165416543952165956343508425568267434014409543695854003881614872032169911128591557214022609814506549519346436666745949765089160103759963579109129494963935148826092199905100687324124377489513527958758552180275987938588826950986857551848595030405060682304853722128110246100956255285132895095654705120185614514543372892864214295920568569533439311335398771177941318039944122071791487484851039109739876915167254039595162138440719064285006933517009969797745712026020405330015588916 [... cut ...] 75174286502221247587974013516691029483658623663513169633783759722257390290840212608024144499509293546557499718602405543855613086920993347433698918308617930654874907280189078021770038762193910060330852670961679528807210088898905136568719256611009516558932692235430795312700357600738529286712640098634811753376972485230297820530547121734188738540991133860565675731390951805218864600647631858209464908206819751430860510055254053440876305103302001953125
```

But, it does crash in xst. We'll take a look.

phoddie, 2021-05-18:
FYI - we have a fix for this that will be included in the next SDK update. The change is isolated to `BigInt.prototype.toString`.

closed 2021-06-03 by phoddie