Moddable-OpenSource/moddable #644: XS: String.fromArrayBuffer and null terminated C-strings

kriskowal, 2021-05-21:
**Description**

`String.fromArrayBuffer` appears to treat the given buffer as a null-terminated C string. This makes it unsuitable as a primitive for the faithful emulation of `TextDecoder`.

**Steps to Reproduce**

Evaluate the expression `String.fromArrayBuffer(new Uint8Array([65, 0, 65]).buffer)`. This returns `"A"`, on account of the null terminated.

**Expected behavior**

The hope is that this would return "\x00A" ("\u0000A") so that it is possible to use `String.fromArrayBuffer` to produce an emulation of the `TextDecoder` present on the web and in Node.js.

```js
$ node
> new TextDecoder().decode(new Uint8Array([0, 65]))
'\x00A'
```

**Work-around**

I can work around this by changing the Zip library I’ve written to use a Uint16 instead of a string for recognizing a compression scheme. A more general solution would be to use a pure JS UTF-8 codec, though that is somewhat superfluous in XS.

phoddie, 2021-05-21:
Handling of nulls in strings is special case in XS. That is for long-standing reasons related to the internal representation being UTF-8 rather than UTF-16 (too much to explain here). The implementation of `String.fromArrayBuffer` doesn't implement special case for nulls in strings , leading to the behavior you see.

There is another issue with `String.fromArrayBuffer` which is a bigger concern: it performs no input validation. That means it it is possible to inject an invalid UTF-8 sequence into an XS string. Code that operates on XS strings assumes they are valid UTF-8, so this is a potential security issue. At a minimum it is currently trivial to force a read a few bytes beyond the end of the string.

Implementing handling of nulls strings and validating that the input is well-formed UTF-8 both require scanning the input prior to the copy.  I implemented a quick patch to do both. It isn't quite proper XS coding style, but it seems to work. ;) The patch for `fx_String_fromArrayBuffer` is [here](https://gist.github.com/phoddie/21c85259e95d840b01edba22c18d7071) if you'd like to give it a try. It wouldn't surprise me if there are some edge cases that aren't correct.

kriskowal, 2021-05-24:
Following from https://github.com/Moddable-OpenSource/moddable/issues/645#issuecomment-846718371 regarding C-strings.

I’m content to see this closed as wontfix/invalid since the current behavior is the right fit for the Moddable community. We have other ways to achieve our ends. We have a mitigation for the motivating case already landed https://github.com/endojs/endo/pull/737, though we’re likely to encounter others. If we can’t faithfully implement `TextDecoder` in terms of `String.fromArrayBuffer`, we can roll our own UTF-8 codec in JS. It’s not onerous, but we will definitely have to fork one since the community is lagging behind embracing ESM and ArrayBuffer.

I have landed a change in our fork of the JSZip https://github.com/endojs/endo/pull/737 so there’s no longer a pressing motivation. The specific case was the treatment of a two byte magic number to indicate the compression strategy, which is '\x00\x00' by default, but otherwise uses uint16s that coincide with a pair of ASCII characters. However, I believe @michaelfig is also using strings as byte slices over JSON with a Cosmos API. We are checking to make sure we’re covered there.

I’m sensitive to the need for Moddable’s API to preserve backward compatibility and appreciate that there’s no pressure for it to conform to any web standard. It would also be a waste to have to follow-up `String.fromArrayBuffer` with `indexOf('\x00')` and `slice` to truncate when interacting with C-FFI.

phoddie, 2021-05-25:
> ...since the community is lagging behind embracing ESM and ArrayBuffer.

The horror!

> The specific case was the treatment of a two byte magic number to indicate the compression strategy, which is '\x00\x00' by default, but otherwise uses uint16s that coincide with a pair of ASCII [...]

If I understand correctly, both of these situations use strings for comparison of binary data, taking advantage of the fact that some binary data happens to be a valid string. That approach is a clever work around for JavaScript's lack of a built-in binary data comparison.  Maybe the language should include that?  (FWIW - the Moddable SDK has a [Bin module](https://github.com/Moddable-OpenSource/moddable/blob/7c584420c905aacf878a08f8a4f3ee99ec0c4135/modules/crypt/bin/bin.js#L38-L39) with `comp`, and I just found `equalArrayBuffer` in another project.)

phoddie, 2021-06-17:
`String.fromArrayBuffer` will continue to stop on a null byte. While it isn't ideal in every situation, it is a useful behavior in many situations. In addition changing the behavior (which we tried) breaks existing code.

closed 2021-06-17 by phoddie

erights, 2021-06-17:
Could we have a variation that does not stop on zero?

phoddie, 2021-06-17:
Yes.

The code is already written in the original revision of this [Gist](https://gist.github.com/phoddie/21c85259e95d840b01edba22c18d7071/revisions). It just needs to be reconciled with the current implementation in XS, which should be very straightforward.

The question is where to put it. I'd rather not merge it into the Moddable SDK default XS as it is kind of obscure and increases the code size a bit. We could just have a small utility module that exports a function -- then you can put wherever you like.

erights, 2021-06-17:
What do you think about packaging these as [TextEncode](https://developer.mozilla.org/en-US/docs/Web/API/TextEncoder) and [TextDecoder](https://developer.mozilla.org/en-US/docs/Web/API/TextDecoder)? There is almost no API there other than these functions. IIUC, as globals, it would fit your framework for including or omitting things by configuration.

phoddie, 2021-06-17:
Nice idea. Yes, we could provide implementations of TextEncoder and TextDecoder. That would avoid creating a new API and provide a some more web compatibility in the Moddable SDK. The implementations would be subset of the full TextEncoder and TextDecoder. I expect you are OK with that as long as it throws when it recognizes an unsupported feature being invoked.

Everything in the Moddable SDK is a module, so the natural way to provide these is as modules. A host can include them as needed and can choose to store those constructors in a global if it wants to provide the same access in the same way as the web. Packaging them as modules also makes it very straightforward to omit functionality unused in a given host. 

```js
import TextEncoder from "text/encoder";
import TextDecoder from "text/decoder";

globalThis.TextEncoder = TextEncoder;
globalThis.TextDecoder = TextDecoder;
```

FWIW - I am generally uncomfortable providing subset API implementations. They can be frustrating for developers already familiar with the full implementations. I don't see a better alternative though -- creating a new API to do about the same thing just to reset expectations doesn't seem developer friendly either.

phoddie, 2021-06-17:
There's quite a bit of functionality hidden in the small `TextDecoder` API:

- Accepts many different text encodings, not just UTF-8
- Has BOM sniffing support (we can requires this be OFF to start)
- Optional streaming decode means supporting decode with a single glyph split across one (or more!) invocations of `decode`
- Appears to allow decoding of malformed input byte streams (`EncodingError` option) by default(!)

None of that is necessary to provide the one feature that started this: allowing embedded nulls in a string.

`TextEncoder` is simpler because the input is always a JavaScript String and the output is always UTF-8.

erights, 2021-06-17:
Gotcha. Yes, omitting all that is fine.

phoddie, 2021-06-21:
FWIW - we should have an initial implementation of this up in the next 24 hours. For API compatibility, it implements `fatal` and `ignoreBOM` in the same way as the Web specs. No doubt there will be other behaviors to refine for compatibility, so this is will just be a start.

kriskowal, 2021-06-21:
This is incredible. Thank you.

I will certainly need some guidance on how to incorporate the text modules into xsnap.

phoddie, 2021-06-21:
Yea, I was thinking it could be an interesting little adventure to get a module integrated into xsnap. It seems like a worthwhile one though, as there will likely be other reason to want to do that.