Moddable-OpenSource/moddable #645: XS: String.fromArray

kriskowal, 2021-05-21:
**Describe**

`String.fromArrayBuffer` is less useful than a hypothetical `String.fromUint8Array` or an overloaded `String.fromArray` because buffers do not support offset views. Supposing one is writing a tool that receives a large buffer containing an encoded file, creates a Uint8Array view of that buffer, then attempts to extract the text within a subarray, it is necessary to allocate a fresh ArrayBuffer of the desired size and copy the view.

**Why do you think this feature would be useful?**

This would improve the performance of applications that carve file formats, like a JavaScript Zip archive reader.

**Describe alternatives you've considered**

The more general solution would be to implement the experimental `TextDecoder` API, which is suitable for some stateful streaming applications and other codecs. I happen to also like the brevity of the XS `String.fromArrayBuffer` API for the common case.

phoddie, 2021-05-21:
Interesting thought. A few comments:

- There's no need for another API, `String.fromArrayBuffer` could safely accept a `Uint8Array`. Granted, the name is not entirely obvious though. I'd hesitate to support TypedArrays in general, as creating a string from a 32-bit floating point array, for example, doesn't seem to have an obviously useful meaning.
- Rather than using a TypedArray,  a `DataView` is an interesting alternative. It really just specifies a window on a buffer without giving any interpretation to the content.
- There was a [proposal](https://github.com/Jack-Works/proposal-arraybuffer-fixed-view#fixed-view-of-arraybuffer) at the last TC39 meeting which, if I understand it correctly, allows for an ArrayBuffer to be a view on another ArrayBuffer. I'm not sure I see a strong need, and it does create some implementation complexity. If added to the language, it would address your idea here.
- `TextDecoder` would be nice, but the generality opens up a bunch of issues that make its implementation non-trivial. It could absolutely be implemented as a module for XS. Having that would be helpful for bringing in code authored for the web (for some quick and dirty porting purposes we've implemented `TextDecoder` using `String.fromArrayBuffer`, but that's obviously limited.) `String.fromArrayBuffer` is a good fit for our embedded needs and has a tiny implementation.

kriskowal, 2021-05-21:
With Agoric-SDK, we do create an emulation of `TextDecoder` and `TextEncoder` so we can run some libraries portably. I’m inspired to post this issue because we forgot to coerce Uint8Array to ArrayBuffer in the version on trunk and I’m fixing that now. It requires entraining a sad memcopy. ArrayBuffer windowing would be nice. We also get some use of `DataView`, though it lacks a zero-alloc view function like `subarray`.

Overloading `String.fromArrayBuffer` so it can accept `ArrayBuffer` and `Uint8Array` is at least consistent with the domain of `TextDecoder().decode()` so that sounds great to me. For example:

```js
String.fromArrayBuffer(new Uint8Array([0, 65, 0]).subarray(1)) === 'A\x00'
```

phoddie, 2021-05-21:
I should have been more clear about that while `String.fromArrayBuffer` could be extended to support `Uint8Array`, I'm not sure it is the best approach. It feels ad-hoc and partial because it would only be useful for a single kind of TypedArray, and even that is really just to have a view.

I think `DataView` makes more sense. The `DataView` constructor supports offset and length arguments, so it is possible to construct a "zero-alloc" subview:

```
let view = new DataView(new ArrayBuffer(128));
let secondHalf = new DataView(view, 64, 64);
```

Using the `byteOffset` property of a `DataView` instance, subviews of subviews can work like subarray.

```
DataView.prototype.subview = function(byteOffset, byteLength) {
   return new DataView(this.buffer, this.byteOffset + byteOffset, byteLength);
}

let lastQuarter = secondHalf.subview(32, 32);
```

kriskowal, 2021-05-21:
It is neat that a `DataView` can be used in this way and it seems that `TextDecoder` overloads for `DataView` as well as typed arrays and `ArrayBuffer`, so it would be possible to refactor some Agoric libraries to treat `DataView` as the common vessel for byte slices. If TC-39 bought `DataView.prototype.subview`, it would even be ergonomic.

kriskowal, 2021-05-21:
I’ve also noticed that the web API’s strongly hint at `Uint8Array` as the vessel-of-choice for arbitrary bytes. `new TextEncoder().encode('hi')` returns a `Uint8Array`, so the encoder and decoder are dual for that content type.

kriskowal, 2021-05-21:
Here’s another alternative approach: if `String.fromArrayBuffer` were extended to accept `String.fromArrayBuffer(buffer, byteOffset, byteLength)`, it would be possible to implement `TextDecoder` without an intermediate buffer allocation. This would be sufficient and not require a type overload.

phoddie, 2021-05-24:
Passing the start and end offset directly is nice and avoids views entirely. That seems like the simplest solution.

I did run into a problem with allowing null bytes. It is pretty common to have binary data structures containing strings that are terminated with a null byte, and even padded with a null bytes to fill out a fixed size field. One such example is the [StateLabel](https://lan.developer.lifx.com/docs/querying-the-device-for-data#statelabel---packet-25) message sent by a LIFX bulb to indicate its name. Using `String.fromArrayBuffer()` on the full label field returns a string with the label padded by nulls. But what is wanted is the string without any training nulls.

So.... the question is what `String.fromArrayBuffer` should do with nulls - keep the original behavior of stopping on a null, change the behavior to return all nulls, or some different behavior  / caller option. I'm not a huge fan of nulls in strings. It is allowed in JavaScript but it isn't exactly usual (surely not in C!). This issue began because you are reading part of the ZIP `archive` into a JavaScript string and that could contain a null. I'm wondering if that case, which isn't even really a string, should guide us here.

phoddie, 2021-05-24:
FWIW - [here's](https://gist.github.com/phoddie/21c85259e95d840b01edba22c18d7071) an updated implementation of `String.fromArrayBuffer` that supports optional `offset` and `byteLength` arguments. Support for nulls is commented out.

kriskowal, 2021-05-24:
I think regardless of the null-terminator issue #644, adding start and end offset arguments would be useful. As I discuss at https://github.com/Moddable-OpenSource/moddable/issues/644#issuecomment-846739909, we have other options for making a faithful emulation of `TextDecoder` if necessary. We’d obviously prefer to do less in JS. At the peril of falling into the final boolean argument trap, perhaps `String.fromArrayBuffer` can be in some way parameterized to conditionally respect a null terminator.

phoddie, 2021-05-25:
I agree that the start/end offsets are separate from the handling of null bytes.

kriskowal, 2021-06-04:
I saw this was addressed in the most recent release notes!

closed 2021-06-04 by kriskowal