Moddable-OpenSource/moddable #647: Including OneWire DS18x20

mauroForlimpopoli, 2021-05-30:
I'm trying to add the driver OneWire of DS18X20, but when I add it in my manifest i get this error:
////////
C:\moddable\examples\piu\k9>mcconfig -d -m -r 90 -p esp32/moddable_two
# bles2gatt bleservices
# xsl modules
### 1243 instances, 1857 keys, 95 colors, 0 holes
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() ShiftTexture() it: not frozen
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() BackspaceTexture() it: not frozen
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() NumKeysTexture() it: not frozen
# cc mc.xs.o (slots in flash)
# mcrez resources
Total resource size: 145663 bytes
# cc mc.resources.o (slots in flash)
# ld xs_esp32.bin
        1 file copiati.
Executing action: all (aliases: build)
Running ninja in directory c:\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable xs_esp32.elf
FAILED: xs_esp32.elf
cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\XTENSA~1\ESP-20~1.0\XTENSA~1\bin\XT1FE9~1.EXE  -mlongcalls -Wno-frame-address   @CMakeFiles\xs_esp32.elf.rsp  -o xs_esp32.elf  && cd ."
c:/progra~1/espres~1/esp-id~1/tools/xtensa~1/esp-20~1.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: xs_esp32.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/progra~1/espres~1/esp-id~1/tools/xtensa~1/esp-20~1.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/progra~1/espres~1/esp-id~1/tools/xtensa~1/esp-20~1.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 768 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
NMAKE : fatal error U1077: 'cd' : codice restituito '0x2'
Stop.

C:\moddable\examples\piu\k9>
////////////
This is my complete manifest:
{	
	"build": {
		"PARTITIONS_FILE": "./partitions.csv"
	},
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_piu.json",
		"$(MODULES)/input/keyboard/manifest.json",
		"$(MODULES)/pins/analog/manifest.json",
		"$(MODDABLE)/modules/files/file/manifest.json",
		"$(MODULES)/network/wifi/manifest.json",
		"$(MODDABLE)/examples/manifest_net.json",
		"$(MODDABLE)/modules/network/ble/manifest_server.json",
		"$(MODULES)/network/mdns/manifest.json",
		"$(MODDABLE)/modules/io/manifest.json",
		"$(MODULES)/pins/digital/manifest.json"
	],
	"creation":{
		"stack":500,
		 "static": 98304,
		 "keys": {
			"available":50
		},
		"chunk": {
			"initial":35362,
			"incremental":1024
		}
	},
	"modules": {
		"*": [
			"./main",
			"$(MODULES)/network/http/*",
			"./restart",
			"./deepSleep",
			"./funcSleep",
			"$(MODULES)/network/sntp/*",
			"$(MODULES)/drivers/onewire/*"
		],
		"esp32/ota": "$(MODDABLE)/build/devices/esp32/modules/ota/*"
	},
	"resources":{
		"*-alpha": [
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Regular-20",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-18",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Light-42",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-28",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Regular-72",
			"$(MODDABLE)/examples/assets/fonts/arc64",
			"$(MODDABLE)/examples/assets/fonts/roboto32"
		],
		"*-image(50)": [
			"$(MODDABLE)/examples/assets/images/Bluetooth-icon",
			"$(MODDABLE)/examples/assets/images/Wifi-icon",
			"$(MODDABLE)/examples/assets/images/unlock-icon",
			"$(MODDABLE)/examples/assets/images/lock-icon",
			"$(MODDABLE)/examples/assets/images/down-icon",
			"$(MODDABLE)/examples/assets/images/battery-icon"
		],
		"*-mask": [
			"$(MODDABLE)/examples/assets/images/loading-icon",
			"$(MODDABLE)/examples/assets/images/thermometer-mask",
			"$(MODDABLE)/examples/assets/images/battery-mask"		
		]
	},
	"platforms": {
		"esp32": {
			"config": {
				"rotation": 90,
				"brightness": 0,
				"file": {
					"root": "/mod/"
				},
				"onewire": {
					"pin": "26"
				}
			},
			"defines": {
				"onewire": {
					"driver_rmt": "RMT",
					"rmt_rx_channel": "RMT_CHANNEL_2",
					"rmt_tx_channel": "RMT_CHANNEL_3"
				}
			},
			"modules": {
				"*": [
				"$(MODULES)/files/flash/*",
				"$(MODULES)/files/flash/esp32/*",
				"$(MODULES)/drivers/onewire/esp/*"
				]
			}
		}
	},
	"preload": [
		"neopixel",
		"http",
		"sntp",
		"file",
		"pins/analog",
		"timer",
		"esp32/ota",
		"ble",
		"preference",
		"flash",
		"Resource",
		"keyboard",
		"time",
		"timer",
		"onewire",
		"DS18X20"
	],
	 "data": {
        "*": [
            "./mydata",
			"./myWeb"			
        ]
    },"ble":{
		"*": [
			"./bleservices/*"
		]
	}
}
What can I do to resolve it ?
Regards

wilberforce, 2021-05-30:
This isn't adding onewire - it's an issue with the total size of your firmware:

> c:/progra1/espres1/esp-id1/tools/xtensa1/esp-201.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 768 bytes

Try decreasing this:
"static": 98304,

try 65535, or even smaller - See if you can get rid of the overflow.

phoddie, 2021-05-31:
The problem is that your project is overflowing IRAM memory (as @wilberforce notes above):

```
'.iram0.text' will not fit in region `iram0_0_seg'
IRAM0 segment data does not fit.
region `iram0_0_seg' overflowed by 768 bytes
```

That's unusual. In most problems the a good chunk of IRAM is unused on ESP32 projects (because it isn't general purpose memory). Changing the static allocation in the manifest won't help, as the JavaScript virtual machine is allocated from general purpose memory, so it can't use IRAM.

Looking at the link map (xs_esp32.map), iram0 is here:

```
Name             Origin             Length             Attributes
iram0_0_seg      0x0000000040080000 0x0000000000020000 xr
```

That's 128 KB. A build of helloworld uses only about 40 KB (41240 bytes).

In the Moddable SDK build, using the `ICACHE_RAM_ATTR` attribute with native code puts the corresponding function / structure in IRAM. Are you using that in any of your native code or otherwise explicitly tagging static allocated structures into IRAM?

mauroForlimpopoli, 2021-05-31:
This is correct in my map.
Name             Origin             Length             Attributes
iram0_0_seg      0x0000000040080000 0x0000000000020000 xr

The error appears when I add this line to the manifest above:
"$(MODULES)/drivers/onewire/*"

attached the file of my native functions
[native.txt](https://github.com/Moddable-OpenSource/moddable/files/6569556/native.txt)

wilberforce, 2021-06-01:
Searching for `ICACHE_RAM_ATTR`
https://github.com/Moddable-OpenSource/moddable/search?p=1&q=ICACHE_RAM_ATTR

There are no references in the onewire source files for this.

How many references to `ICACHE_RAM_ATTR` do you have in your native code - and what is the total length - does it exceed 128KB ?

phoddie, 2021-06-01:
> This is correct in my map.
>> Name Origin Length Attributes
>> iram0_0_seg 0x0000000040080000 0x0000000000020000 xr

The address and length are based on the hardware, so they should be the same in every ESP32 link map. I included it as a reference to show the size of the `IRAM0` block.

The only practical way to sort out what is going on is to look at the link map to see what is overflowing the `IRAM0` block. If you aren't sure how to do that, I can take a look. Please find the `xs_esp32.map` file for your project (at a path like `$MODDABLE/build/bin/esp32/moddable_two/debug/YOUR_PROJECT_NAME/xs_esp32.map`). Add a `.txt` extension to the file (e.g. `xs_esp32.map.txt`). Attach the file `xs_esp32.map.txt` to your GitHub reply by drag and drop. Include a link to the attached file like [this](https://github.com/Moddable-OpenSource/moddable/files/6573625/xs_esp32.map.txt).

mauroForlimpopoli, 2021-06-01:
This is my map
[xs_esp32.map.txt](https://github.com/Moddable-OpenSource/moddable/files/6575757/xs_esp32.map.txt)

phoddie, 2021-06-01:
Interesting. `IRAM0` is really full and there's no obvious single block or source file to blame. In a project that enables both BLE and Wi-Fi, IRAM0 gets up to about 127 KB of the possible 128 KB. Adding much else pushes it over the edge.

In reviewing the link map, much of the space is taken up by Wi-Fi related code. There are are options in the SDKCONFIG to move parts of the Wi-Fi stack into IRAM for to improve Wi-Fi throughput. The Moddable SDK does not enable those in its ESP32 [sdkconfig.defaults](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj-esp32/sdkconfig.defaults). It looks like the ESP32-IDF defaults some of those IRAM optimizations to enabled.

In your project's `sdkconfig.defaults` (or in the MODDABLE SDK `sdkconfig.defaults`, if you don't have one in your project yet), you adding these two lines takes IRAM usage down to under 96 KB when Wi-Fi and BLE are both enabled.

```
CONFIG_ESP32_WIFI_IRAM_OPT=n
CONFIG_ESP32_WIFI_RX_IRAM_OPT=n
```

mauroForlimpopoli, 2021-06-02:
I followed your indications and now all it works correctly.
Than you Peter

closed 2021-06-02 by phoddie

mauroForlimpopoli, 2021-06-07:
Now I'm testing the commands and all the example run correctly. But how can I write the scratchpad ?
To read it I used:
devices.scratchpad = new ArrayBuffer(9);
let scrshorts = new Uint8Array(devices.scratchpad);
 trace("scr Before:", scrshorts[0], '\n');
  trace("scr Before:", scrshorts[1], '\n');
  trace("scr Before:", scrshorts[2], '\n');
is it right ?

phoddie, 2021-06-08:
1. The original is closed as resolved. This is a different topic that happens to reference the same source file. It does't belong in this issue.
2. This appears to be a question, not an issue. If that is correct, please use Discussions.
3. The implementation of DS18x20 clearly marks the scratchpad accessors as "internal use" so it may be that they were not intended to be called directly.