Moddable-OpenSource/moddable #650: GIPHY app stops when search response data:[]

stc1988, 2021-06-05:
**Build environment:** Windows
**Target device:** M5stack core2

**Steps to Reproduce**
1. GIPHY app launch with `mcconfig -d -m -p esp32/m5stack_core2`
2. touch OK button of keyboard. Search string is empty.
3. See error

```
Searching for  gifs...
moddable\contributed\giphy\main.js (53) # Break: ?: cannot coerce undefined to object!
```
response value is 
```
{
  "data": [],
  "pagination": {
    "total_count": 0,
    "count": 0,
    "offset": 0
  },
  "meta": {
    "status": 200,
    "msg": "OK",
    "response_id": "API_KEY"
  }
}
```

phoddie, 2021-06-05:
Did you follow the instructions in the GIPHY apps [readme.md](https://github.com/Moddable-OpenSource/moddable/blob/public/contributed/giphy/readme.md) to add your own GIPHY API key to the app's manifest?

stc1988, 2021-06-05:
Yes, I can work with my GIPHY API key.

`"response_id": "API_KEY"`
Sorry, it's confusing.  I  replace  my API key  with 'API_KEY' to hide my API key.

lprader, 2021-06-17:
Thanks for the report. I've committed a fix for this so the app will display an error message when no GIFs are found for a search query. The fix will be merged into the next Moddable SDK update.

stc1988, 2021-06-21:
@lprader 
I confirmed 2a47b79aaffb7331835b01eaf6315ac0b638ba4d and it works as expected.
Thanks.

closed 2021-06-21 by stc1988