Moddable-OpenSource/moddable #651: Correct way to update to ESP IDF 4.21

wilberforce, 2021-06-07:
I have done:

- cd to esp-df folder
- git fetch
- git checkout v4.2.1
- git submodule update --init --recursive

however I now see:
```
> git status

HEAD detached at v4.2.1
nothing to commit, working tree clean
```

Is there a way to be at this release without a detached head?

phoddie, 2021-06-08:
In the "worked for me" category: on macOS, I just followed the instructions and all worked well. 

One note: not doing the `source $IDF_PATH/export.sh` step on macOS leaves things in a bad state. That doesn't seem to be part of the Windows process though.

I respectfully defer to @andycarle.

andycarle, 2021-06-08:
> I have done:
> 
> * cd to esp-df folder
> * git fetch
> * git checkout v4.2.1
> * git submodule update --init --recursive
> 
> however I now see:
> 
> ```
> > git status
> 
> HEAD detached at v4.2.1
> nothing to commit, working tree clean
> ```
> 

You don't actually want to do `git checkout v4.2.1`.  Doing that checks out the reference `v4.2.1` which is a tag and will leave you in a detached head state, as you note.

Instead, you should still be on the `release/v4.2` branch. Espressif has just tagged that branch (confusingly) with `v4.2.1` now.

As detailed in [our ESP32 documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#updating-1) the right sequence is still:

```
cd $IDF_PATH
git fetch
git checkout release/v4.2
git submodule update --init --recursive
git pull
```

> cd %MODDABLE%
> git fetch --all --tags
> git checkout  OS210603
> ```
> 
> I now have:
> 
> ```
> >git status
> HEAD detached at OS210603
> ```
> 
> Is there a way to do this with the detached heads?

You don't need to check out a tag here either, unless you want to. Just doing:

```
git checkout public
git pull
```

should be fine. The `public` branch is always the most up-to-date version of the repository. The tags just match with specific releases and release notes.

wilberforce, 2021-06-08:
Thanks.

The thing that doesn't make much sense is this:
`git checkout release/v4.2`

when in fact its now aliasing v4.2.1

andycarle, 2021-06-09:
Agreed; I don't love that Espressif went with that convention. But, unfortunately, there's not much we can do on our end other than document the oddity.

I'll add a small note to our ESP32 doc to this effect.

phoddie, 2021-06-17:
Is this resolved or is there something more that remains to do?

andycarle, 2021-06-17:
There's a commit pending in our tree (c4248f054555feac242664ebd2f8a9015e2e3d98) that is intended to address this. I'll close it out when that commit goes live on GitHub.

closed 2021-06-18 by wilberforce