Moddable-OpenSource/moddable #652: bigint string form is codepoint rather than numeral in OS210603

dckc, 2021-06-08:
**Steps to Reproduce**
1. `$ git checkout b80d5c675ebc52b7cb6c098baa276ac736224f4d; cd xs/makefiles/lin; make`
2. `$ $MODDABLE/build/bin/lin/release/xst -e 'print(65n)'`
3. `A` is printed

likewise, `97n` prints as `a`. `1n` prints as a control character.

**Expected behavior**
`65` is printed (`97` resp)

**Other information**
I'm pretty sure this broke somewhere in e70607c4..edf1ddf4 . I see some changes to `fxBigintToString`. I'm not sure what to make of them.

context: https://github.com/Agoric/agoric-sdk/pull/3262#issuecomment-856358427
cc @warner @phoddie

dckc, 2021-06-08:
I wonder if/how this made it thru test262.

```
xst -e 'function assertEquals(a, b) { if (a !== b) throw Error("not equal") }; assertEquals("297019028", BigInt(0x11b42694).toString(10))'
Error: not equal
```

https://github.com/tc39/test262/blob/6e61dd7754e7c94ebcf3ee679cb8db3c54a37b50/implementation-contributed/v8/mjsunit/harmony/bigint/basics.js#L144

phoddie, 2021-06-08:
Yea, I thought the same thing when I read your original report. I think we may have pushed an incomplete commit, as it is working for me with `xst` this morning with our internal repository. 

```
$ xst -e 'print(`${1n}` === "1")'
true
```

Will sort it out. Thanks for the report.

phoddie, 2021-06-08:
It looks like the [fix](https://github.com/Moddable-OpenSource/moddable/commit/b63642cdc4e460d061199e4a74ebedb749cd006e) for that got pushed yesterday. ;)

dckc, 2021-06-09:
fix confirmed.

closed 2021-06-09 by dckc