Moddable-OpenSource/moddable #662: ESP32 debugger not working

gidagel, 2021-07-01:
**Build environment:** Linux Ubuntu 
**Target device:** ESP32

**Description**
Hi, I have an ESP32 and I'm being able to program it in Arduino normally. However, I installed Moddable on Ubuntu, it compiles, writes on the board, but apparently it doesn't work (I tested it with Blink Example and the LEDs didn't blink), also nothing appears in the debugger. Below is the screenshot of the current state. 

**Expected behavior**
I expected the board to blink and to see the xsdebug working.

**Images**
[image]

mkellner, 2021-07-01:
I just did a fresh install on Ubuntu 20.04.1 without problems.

Ensure you are up to date with ESP IDF. Here is what my install looked like:

[image]

Follow the install instructions for linux including the
`cd $IDF_PATH
./install.sh`

What was the build line that you used? Did the platform (`-p` option) match your device?

I did a build for a Moddable Two
`mcconfig -d -m -p esp32/moddable_two`
and for a raw esp32 chip with
`mcconfig -d -m -p esp32/nodemcu`
or
`mcconfig -d -m -p esp32`

gidagel, 2021-08-13:
Hi,

I reinstalled with v.4.2.1 and it worked.

Thank you for your help!

closed 2021-08-13 by gidagel