Moddable-OpenSource/moddable #665: super-Latin1 white space after regular expression literal is incorrectly rejected

gibson042, 2021-07-14:
**Environment:** XS 10.5.0

**Description**
In the ECMAScript lexical grammar, [regular expression literal](https://tc39.es/ecma262/#prod-RegularExpressionLiteral) flags are defined as a sequence of [_IdentifierPart_](https://tc39.es/ecma262/#prod-IdentifierPart), each of which is either an ASCII dollar sign `$`, zero-width joiner or non-joiner, `\u…` escape sequence, or code point with the Unicode property “ID_Continue”. However, _WhiteSpace_ characters with code points above the Latin-1 Supplement block (none of which are matched by _IdentifierPart_) after _RegularExpressionFlags_ are incorrectly rejected rather than ignored as insignificant.

**Steps to Reproduce**
1. Evaluate source text like `/x/ ` (i.e., a regular expression literal followed by U+1680 OGHAM SPACE MARK).
2. Check all _WhiteSpace_ and _LineTerminator_ characters (as identified at [unicode.org](https://util.unicode.org/UnicodeJsps/list-unicodeset.jsp?a=%5B%5B%5B%3ASpace_Separator%3A%5D%5B%5Cu000A%5Cu000D%5Cu2028%5Cu2029%5D%5D%26%5B%5E%3AID_Continue%3A%5D%5D&g=&i=)):
   ```
   ["000A", "000D", "0020", "00A0", "1680",
    "2000", "2001", "2002", "2003", "2004", "2005", "2006", "2007", "2008", "2009", "200A",
    "2028", "2029", "202F", "205F", "3000"
   ].filter(hex => {
     const cp = String.fromCharCode(parseInt(hex, 16));
     try {
       eval("/x/" + cp);
       // No parsing error, suppress code point.
       return false;
     } catch ( ex ) {
       // Parsing error, keep code point.
       return true;
     }
   })
   ```

**Expected behavior**
1. No error.
2. An empty array.

**Actual behavior**
1. _SyntaxError: invalid character 5760_
2. 1680,2000,2001,2002,2003,2004,2005,2006,2007,2008,2009,200A,202F,205F,3000

**Script**
```sh
$ ./xs -v; cat /tmp/js; ./xs /tmp/js
XS 10.5.0
const ws = ["000A", "000D", "0020", "00A0", "1680",
  "2000", "2001", "2002", "2003", "2004", "2005", "2006", "2007", "2008", "2009", "200A",
  "2028", "2029", "202F", "205F", "3000"
];
const rejected = ws.filter(hex => {
  const cp = String.fromCharCode(parseInt(hex, 16));
  try {
    eval("/x/" + cp);
    // No parsing error, suppress code point.
    return false;
  } catch ( ex ) {
    // Parsing error, keep code point.
    return true;
  }
});
print("rejected: " + rejected);
rejected: 1680,2000,2001,2002,2003,2004,2005,2006,2007,2008,2009,200A,202F,205F,3000
```

gibson042, 2021-07-14:
Updated to reflect that this issue is more general than the [regular-expression-specific one in JavaScriptCore](https://bugs.webkit.org/show_bug.cgi?id=227944).

phoddie, 2021-07-28:
Thank you for the detailed report.  You are absolutely correct that the problem was more general than just `RegExp`.

This is fixed in today's Moddable SDK update. The `xst` binary hasn't been updated yet, so to verify the fix you need to build from sources.

closed 2021-08-28 by phoddie