Moddable-OpenSource/moddable #669: Touch y axis reversed in Ch10 examples with Moddable two board

mulrich, 2021-07-20:
Hi I am new to this environment and have been running through examples on a Moddable Two board.  When I run any of the Ch10 examples: rounded, buttons, keyboard, etc. the touch does not line up with piu object on the y axis. i.e. clicking on option 3 button highlights option 2 button and hitting lower left below option 3 highlights option 1 button on the top. It behaves as if the origin for the touch is in the lower left and origin for piu is upper left.  I have not done anything I am aware of to change or adjust anything in the default install of release 4.2. Other non-touch examples all working fine. Tried a clean build and two different moddable two boards with same behavior. Tried with simulator on the Mac and it behaves properly so appears to be moddable two board specific. Please advise on anything I may have missed on setup or configuration.

andycarle, 2021-07-20:
Curious! That sounds like a hardware fault of some sort, but you may be able to work around it in software. 

The change you can try is to Line 43 of [`$MODDABLE/build/devices/esp32/targets/moddable_two/manifest.json`](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/moddable_two/manifest.json#L43).

 There you'll see `"flipY": false,`. Try changing that to `"flipY": true,` and then rebuild/deploy the ch10 host and then the example. If that gives unexpected results, try changing `flipX` instead.

Please let me know if that shows any sign of improvement for you. If not, we'll explore other ways to get you back up and running.

mulrich, 2021-07-21:
Thanks Andy.   Actually had to change flipX = true and flipY = false and that solved the orientation issue!  Verified examples now run properly. BTW this same behavior was on two Moddable two different moddable two boards.

closed 2021-07-21 by mulrich