Moddable-OpenSource/moddable #674: Bug: Number.prototype.toLocaleString complains about radix

mathiasbynens, 2021-07-27:
```
$ xs -e 'print(42..toLocaleString("en"))'
RangeError: Number.prototype.toLocaleString: invalid radix
```

The expected result is the string `"42"`.

Note that `Number#toLocaleString` doesn’t deal with radix so aside from the fact no error should be thrown at all, the error message itself seems incorrect — perhaps there is a more fundamental issue here?

phoddie, 2021-07-27:
Hey, @mathiasbynens, thanks for the report. Good catch. Yea, it is more fundamental -- the implementation of `Number#toLocaleString` is just an alias to `Number#toString`. That's just wrong. We'll fix that.

FWIW - the spec text for `Number#toLocaleString` is perhaps slightly misleading:

> This function is implementation-defined, and it is permissible, but not encouraged, for it to return the same thing as `toString`.

What it doesn't say is that it should return the same thing as `toString` **called with no arguments**.

Funny that this error isn't reported by test262. While there is some implementation defined flexibly, throwing an exception on valid input likely wasn't intended to be allowed.

mathiasbynens, 2021-07-27:
Good catch. It would be nice to clarify this in the spec + send a Test262 PR.

phoddie, 2021-07-28:
This is fixed in today's Moddable SDK update. The `xst` binary hasn't been updated yet, so to verify the fix you need to build from sources.

closed 2021-08-28 by phoddie