Moddable-OpenSource/moddable #678: correctness issue: `new BigInt64Array(new Int32Array(0))` should throw

bakkot, 2021-08-04:
Run `new BigInt64Array( new Int32Array(0) )`. This should throw. per https://tc39.es/ecma262/multipage/indexed-collections.html#sec-initializetypedarrayfromtypedarray step 17.c.

Instead, it succeeds, creating a length-0 BigInt64Array.

The same issue happens when going the other direction; that is, `new Int32Array(new BigInt64Array(0))` also incorrectly fails to throw. (Ditto for all other Big/non-Big TypedArray pairs.)

It (correctly) throws when the length of the converted-from array is at least 1.

(V8 has [the same issue](https://bugs.chromium.org/p/v8/issues/detail?id=12052). Firefox correctly implements the spec.)

bakkot, 2021-08-04:
Similarly, `new Int32Array(1).set(new BigInt64Array(0))` incorrectly fails to throw. See [SetTypedArrayFromTypedArray](https://tc39.es/ecma262/multipage/indexed-collections.html#sec-settypedarrayfromtypedarray) step 18.

phoddie, 2021-08-04:
Hey, @bakkot, thanks for the report. Looks like Safari also has the same behavior (not using BigInt64 though). XS and V8 are the same, as you note.

We can fix this in XS. Before diving into that, do you think there is any web-compat issue lurking here that might cause web reality to win out? I suppose that Firefox gets it right makes that less likely.

bakkot, 2021-08-04:
> Looks like Safari also has the same behavior (not using BigInt64 though)

Safari doesn't implement BigInt typed arrays, at least on my machine, so I'm not sure what you mean here: this issue is specifically about constructing a BigInt typed array from a non-BigInt typed array, or conversely, and as such is not observable without BigInt typed arrays.

> do you think there is any web-compat issue lurking here that might cause web reality to win out

Not really - it's quite a new feature, and as you say, Firefox implementing the spec correctly means that such pages were already broken. But I think more importantly, this is only observable with length-0 typed arrays - if you were passing an array of length greater than 0, you would already get an error. And it's hard to imagine someone passing a length-0 array except as a case of passing a length-N array where N happens to be 0, so they'd already be exposed to this error.

phoddie, 2021-08-04:
Regarding Safari, my mistake. I tried a few other tests, not all using BigInt64. Clearly my intuitions about the behavior of TypedArray constructors clearly differ from what is spec-ed.

We'll move ahead following the spec and Firefox example.

closed 2021-10-04 by phoddie