Moddable-OpenSource/moddable #679: print() cannot handle Symbols

warner, 2021-08-06:
We have some error-reporting code which tries to use the built-in `print()` function, and today it was asked to print a symbol. That causes `print` to throw an error:

```js
try {
  print('print registered:', Symbol.for('registered')); // fails
} catch(e) {
  print('error', e);
}
```

results in:

```
print registered: error TypeError: Cannot coerce symbol to string
```

The same happens for `Symbol('anonymous')`, which is how you get the other kind of Symbol.

It would be lovely if `print()` could do the same thing that Node's `console.log` does, which emits `Symbol(registered)`.

I note that `Symbol.for('foo').toString()` returns `Symbol(foo)`, for both kinds of symbols on both XS and Node. But:

```js
const s = `${Symbol.for('foo')}`;
```

throws a TypeError, with a message of `Cannot convert Symbol value to string` on Node, and `Cannot coerce Symbol to string` on XS. I've never understood why the default template literal function isn't using `.toString` on its `${}` arguments, but I imagine there something in the spec about it, so I wouldn't expect XS to behave any differently. But it'd be awfully handy if `print` could do `.toString()` on its arguments.

This is using the XS library as of commit 14e252e82e8c483259f3fe4213e6057aa27b79ba , on Linux, using the `xsnap` environment.

thanks,
 -Brian

phoddie, 2021-08-06:
The global `print` function isn't built into XS. It is provided by the host. In both the `xst` and `xsnap` hosts the implementation of `fx_print` looks to call `fxToString` (directly or indirectly) on the arguments. But, that throws as you note. So... there would need to be some host code in the implementation of `print` to special case Symbol arguments. That should be doable.

phoddie, 2021-08-28:
Coming back around to this, the implementation of `print` uses `fxToString` to coerce the arguments to a string. `fxToString` implements the [`ToString` abstraction operation](https://tc39.es/ecma262/#sec-tostring) defined in Ecma-262. We can't change `fxToString` but we don't have to always use it in `print`.

At the top of `xs_print` in xsnap, there's a loop to coerce the arguments to strings. The version below makes a special case for symbols by calling the `toString` method instead. That seems to give the intended result. What do you think?

```js
void xs_print(xsMachine* the)
{
	xsIntegerValue c = xsToInteger(xsArgc), i;
	for (i = 0; i < c; i++) {
		if (xsSymbolType == xsTypeOf(xsArg(i)))
			xsArg(i) = xsCall0(xsArg(i), xsID("toString"));
		else
			xsToString(xsArg(i));
	}
	...
```

erights, 2021-08-28:
Better would be to do the equivalent of the JavaScript `String(val)` coercion
https://tc39.es/ecma262/#sec-string-constructor-string-value
which is identical to the internal ToString on non-symbols, but uses 
https://tc39.es/ecma262/#sec-symboldescriptivestring
rather than `.toString()` on symbols.

erights, 2021-08-28:
```
> Symbol.prototype.toString = () => 'foo';
> sym = Symbol('x');
> String(sym)
'Symbol(x)'
> sym.toString()
'foo'
```

phoddie, 2021-08-28:
@erights, thanks for the suggestion. That nicely reuses the special case already in the language. Here's the change:

```js
void xs_print(xsMachine* the)
{
	xsIntegerValue c = xsToInteger(xsArgc), i;
	xsVars(1);
	xsVar(0) = xsGet(xsGlobal, xsID("String"));
	for (i = 0; i < c; i++)
		xsArg(i) = xsCallFunction1(xsVar(0), xsUndefined, xsArg(i));
```

phoddie, 2021-09-23:
Closing as this has been addressed in xst. The same change can be applied to xsnap.

closed 2021-09-23 by phoddie