Moddable-OpenSource/moddable #684: mcconfig -t deploy doesn't work

gavrilyak, 2021-08-14:
**Build environment:** or Linux
**Target device:**  Moddable Two

**Steps to Reproduce**
1. `cd $MODDABLE/examples/helloworld; mcconfig -d  -m -p esp32/moddable_two -t build`
2. `mcconfig -d -m -p esp32/moddable_two -t deploy`
3. See error 

```/home/gvv/code/moddable/build/tmp/esp32/moddable_two/debug/helloworld/makefile:189: ESP32_SUBCLASS esp32
/home/gvv/code/moddable/build/tmp/esp32/moddable_two/debug/helloworld/makefile:212: Detected ESP-IDF version v4.2.2. Expected ESP-IDF version v4.2.1.
# uploading to esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/gvv/code/moddable/examples/helloworld/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
CMakeLists.txt not found in project directory /home/gvv/code/moddable/examples/helloworld
make: *** [moddable/build/tmp/esp32/moddable_two/debug/helloworld/makefile:548: DEPLOY_START] Error 2
```
**Proposed solution**
Removing `	idf.py set-target $(ESP32_SUBCLASS)` from tools/mcconfig/make.esp32.mk line 389 seems to fix it...

phoddie, 2021-08-15:
@gavrilyak, thanks for the report. We're able to reproduce the problem. The proposed change does resolve the immediate problem.

In reviewing the change, We did note that `DEPLOY_CMD` does not set `ESP32_SUBCLASS` which may be a problem when working with `ESP32-S2` or `ESP32-S3`. The line you commented out may have been to support ESP32 subclass with a previous version of the ESP-IDF. We're checking that.

gavrilyak, 2021-08-16:
There was similar change for target all:
```
commit 01fa84b519bce81b420d59f2d07f83195e23f58d
Author: Andy Carle <andy@moddable.tech>
Date:   Wed Jan 6 22:49:30 2021 -0800

    avoid unnecessary cleans by not using idf.py set-target (setting IDF_TARGET is handled by sdkconfig.defaults)

diff --git a/tools/mcconfig/make.esp32.mk b/tools/mcconfig/make.esp32.mk
index 6481fae0..d94891c3 100644
--- a/tools/mcconfig/make.esp32.mk
+++ b/tools/mcconfig/make.esp32.mk
@@ -384,7 +384,7 @@ SDKCONFIG_H = $(SDKCONFIG_H_DIR)/sdkconfig.h
 all: precursor
        $(KILL_SERIAL_2_XSBUG)
        $(DO_XSBUG)
-       cd $(PROJ_DIR) ; idf.py set-target $(ESP32_SUBCLASS) ; $(BUILD_CMD) || (echo $(BUILD_ERR) && exit 1)
+       cd $(PROJ_DIR) ; $(BUILD_CMD) || (echo $(BUILD_ERR) && exit 1)
```