Moddable-OpenSource/moddable #686: IDE pour développement

pc152, 2021-08-16:
Bonjour, 

Quel éditeur utiliser sous Windows pour développer avec Moddable.

J'ai suivi toute la procédure d'installation sous Windows et j'ai pu faire le test de compilation et du debug.

Mais je ne comprends pas avec quel editeur on peut développer. 

Merci

wilberforce, 2021-08-16:
You can use any editor your want - even notepad!

VScode works very well: https://code.visualstudio.com/download 
<https://code.visualstudio.com/download>

On 17/08/2021 9:33 am, pc152 wrote:
>
> Bonjour,
>
> Quel éditeur utiliser sous Windows pour développer avec Moddable.
>
> J'ai suivi toute la procédure d'installation sous Windows et j'ai pu 
> faire le test de compilation et du debug.
>
> Mais je ne comprends pas avec quel editeur on peut développer.
>
> Merci
>

pc152, 2021-08-17:
Ok merci, le fait d'installer Visual studio community n'est que pour les outils de compilation ?

wilberforce, 2021-08-17:
> Ok merci, le fait d'installer Visual studio community n'est que pour les outils de compilation ?

Yes - for compilation only

This is a complete editor with plugins for javascript:

<https://code.visualstudio.com/download>

phoddie, 2021-08-28:
This appears to be resolved. Closing. Please re-open if needed.

closed 2021-08-28 by phoddie