Moddable-OpenSource/moddable #697: Piu Skin state more than 3?

meganetaaan, 2021-08-30:
Current piu implementation [limits the number of Skin#state property up to three](https://github.com/Moddable-OpenSource/moddable/blame/6d7f33f8f318663bd5ba8d6ca6536443d42e68ea/modules/piu/All/piuSkin.c#L148). Do you have any reason for it? If any, I think this limitation should be documented on [piu.md](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/piu/piu.md)

For my project I was trying to make [an image of emoticon](https://github.com/meganetaaan/moddable-avatar/blob/master/assets/images/emoticon.png) of six states. Each state represents different icon and has animation using `variant`. Should I separate them into independent images?

stc1988, 2022-02-17:
I am also interested in this issue.
@phoddie  could you take a look?

phoddie, 2022-02-18:
Please recall that the skin also contains colors, and those [colors are selected by the state](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/piu/piu.md#colored-skins-and-styles).  From piu.md:

> In dictionaries, colors can be a single color or an array of 2, 3, or 4 colors. The skin and style objects blend a color based on the state of the content object that uses them.

The `PiuSkinStruct` reserves space for [up to 4 colors](https://github.com/Moddable-OpenSource/moddable/blob/d085d3a61b67cc50cb4f1848bd0754373319972c/modules/piu/All/piuAll.h#L211-L228). Consequently, the code to draw a skin pins the state value to be between 0 and 3 (inclusive).

It is possible to increase the colors array to hold more entries, but that will also increase memory use unnecessarily for many objects that do not use that space. It is possible to dynamically allocate the colors array, but that is much more complex, so there would need to be a strong motivation.

Maybe there is another solution to the problem you are trying to solve? From the description and example image provided, I'm not sure I understand the problem clearly yet. If are willing to explain a bit more, perhaps someone will have a suggestion.

phoddie, 2022-02-24:
@meganetaaan and @stc1988 – do you have any feedback on this? If not, I think we should close this issue.

meganetaaan, 2022-02-26:
Now I understand the specs above. I agree to limit the number of states to save memory. I will separate my emoticon sprite to several image files. (OR I think I got another great option of outline images that enables scaling/rotating! I also gonna try them.)

closed 2022-02-26 by meganetaaan

eltoroit, 2022-02-26:
I faced this issue this morning, I was creating a skin with 10 colors, I wanted to see which one looked better and then reduce it to fewer number of colors.

My problem is that only the first four colors were being shown…. It was clipping from 0 to 3, which makes sense based on the explanation above. But it was not giving me an error or a warning, which would be great to see that in the console.