Moddable-OpenSource/moddable #699: Instrumented build doesn't work on esp32

gavrilyak, 2021-09-04:
**Build environment:** Linux
**Target device:**  Moddable Two
**Steps to Reproduce**
```
$ cd $MODDABLE/examples/helloworld
$ mcconfig -i -m -p esp32
### Error: (host): No such file or directory
```
**Other information**
It seems to work if condition in line 125 of `mcmanifest.js` is changed from
`(tool.debug ? "debug" : "release")`
to
`(tool.debug ? "debug" : tool.instrument? "instrument" : "release")`

but I am to sure whether this is a proper fix

phoddie, 2021-09-04:
Good catch. It does fail with the `esp32` target. We'll review your proposed change (thank you!).

It does work as expected with the `esp32/moddable_two` target:

```
mcconfig -i -m -p esp32/moddable_two
```

gavrilyak, 2021-09-04:
No, it doesn't work with `esp32/moddable_two` target, error is the same. It works after release build though.
It's enough to start `mcconfig -m -p esp32/moddable` and immediately stop it with Ctrl-C, after that instrumented build will work because it will find 'release' directory instead of  'instrument'
But it will break again after `rm  -r build/tmp`

phoddie, 2021-09-06:
My mistake.

Your proposed fix looks correct. We'll get that into a push this week. Thank you again.