Moddable-OpenSource/moddable #705: Support the M5Stack ESP32 Core Ink Development Kit

braechnov, 2021-09-21:
**Describe**
Add build support for the [M5Stack ESP32 Core Ink Development Kit](https://shop.m5stack.com/collections/low-power/products/m5stack-esp32-core-ink-development-kit1-54-elnk-display)

**Why do you think this feature would be useful?**
You already support a range of M5 Stack products, it would surely make sense to extend that range and _please_ support this product as well.

**Describe alternatives you've considered**
None considered yet other than having to use Arduino or ESP-IDF

phoddie, 2021-09-22:
We can take a look. The display driver is likely the most significant part of the effort. We don't have one here. Do you happen to have a spare unit??

braechnov, 2021-09-22:
If you're prepared to take a look I'll buy you one!

phoddie, 2021-09-22:
All in. Send an email to info@moddable.com and I'll send you our office address.

phoddie, 2021-09-23:
(Hardware is on the way!)

phoddie, 2021-10-04:
Hardware arrived. Here's a test pattern drawn using a quick port.

[image]

phoddie, 2021-10-05:
More progress: this image was drawing using Poco. The image shows five levels of gray even though the display has only black and white pixels. The dithering is done in the driver -- Burkes dither here though I intend to switch over to the venerable  [Atkinson dither](https://twitter.com/phoddie/status/1274054350034223104).

[image]

tribalmatt, 2021-10-05:
Hi Peter,

Are these samples in the SDK?  Did I miss them when I was trying to setup
the Heltec OLED?

Thx,
Matt

phoddie, 2021-10-06:
The M5Core Ink port is [available](https://github.com/phoddie/m5core_ink)! It includes an entirely new [display driver](https://github.com/phoddie/m5core_ink/blob/master/targets/m5core_ink/gdew0154m09.js) written using [Ecma-419](https://419.ecma-international.org) IO [APIs](https://419.ecma-international.org/#-9-io-class-pattern) and the new Commodetto [Dither module](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/commodetto/commodettoDither.c). It works well but there's still some bits and pieces to do, so it is in my personal repository for now. The [readme](https://github.com/phoddie/m5core_ink/blob/master/readme.md) explains the details.

phoddie, 2021-12-29:
Closing as support fro M5Core Ink (and M5Paper) have been merged to the Moddable SDK repository. Enjoy.

closed 2021-12-29 by phoddie