Moddable-OpenSource/moddable #706: BLE Security Server and Security Client examples

gitandlucsil, 2021-09-22:
I have a question about the Security Server and Security Client examples, that can be bouth founded in moddable/examples/network/ble folder.

To develop I'm using to ESP32, and one I've flashed the Security Server example and the other one I've the Security Cliente example, and to test and figure out better how authenticate works in the Moddable BLE support, I was chaging the ioCapability attribute inside the securityParameters object, until I both configured Server and Client side this way:

`this.securityParameters = { mitm:true, ioCapability:IOCapability.NoInputNoOutput };`

Doing this, the ESP32 in Client side starts the Scan, find the Server (which is  sending Advertising Packages) and stablish a connection with the Server. But after that, does not happen any authentication and for both sides, never the `onAuthenticated` callback is called.

So, I have this question. When the both devices are configured as `NoInputNoOutput` the Authentication never happen? Because in my mind, it will happen but without any user interaction, with the devices pairing each outher automatically. 

Just for example, when a configure an ESP32 as `BLEServer` and `NoInputNoOutput` in security parameters and make a BLE connection with my mobile phone, on the phone side a pairing request is made, and just after that, the authentication is done.

gitandlucsil, 2021-09-24:
@bfriedkin I saw you had a problem with pairing in BLE using ESP32:

https://github.com/espressif/esp-idf/issues/3532

Could you help me?

lprader, 2021-10-07:
It's not surprising to me that `onAuthenticated` is never called when both of your ESP32 devices are configured as you described and you pair them. With that configuration, the two devices use the "Just Works" pairing method. With the "Just Works" pairing method, there's no authentication procedure, so I wouldn't expect`onAuthenticated` to be called.

(FWIW, here's a good [blog post about Pairing](https://www.bluetooth.com/blog/bluetooth-pairing-part-2-key-generation-methods/) from the Bluetooth website. Step 4 includes a table that says what key generation method is used given different IO capabilities for the two devices being paired.)

It *is* surprising to me that `onAuthenticated` is called when you have one ESP32 device configured as you described and you connect your phone to it. I wouldn't expect it to be called with this setup either. But...I was able to reproduce what you're seeing. The `onAuthenticated` callback is called if I use an app like LightBlue or nRF Connect that shows a confirmation popup when I select a device I want to connect to. After I press "Confirm" or "Pair", `onAuthenticated` is called. If I don't press "Confirm" or "Pair", `onConnected` is called but not `onAuthenticated`.

Looking at the native BLE code, it looks like `onAuthenticated` is a bit of a misnomer here. It's not being called because an authentication procedure has finished; it's called in response to encryption being enabled (as you can see [here](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/ble/nimble/modBLEServer.c#L644)).

So...to answer the question:
> When the both devices are configured as NoInputNoOutput the Authentication never happen? 

That's correct, and that's expected per the Bluetooth spec. But the `onAuthenticated` callback is being called here in response to encryption being enabled, not because an authentication procedure has finished as I would have expected.

gitandlucsil, 2021-10-13:
Thanks @lprader. Another point that I realized, making another tests, when I use the NimBLE stack, the behavior is the same that I reported and I also had in your tests, but, when I use the Bluedroid stack, the `onAuthenticated` method is called when two `NoInputNoOutput` devices connected each other.

phoddie, 2022-01-05:
This doesn't seem to be a Moddable SDK issue, but a consequence of differences in the underlying BLE implementation.

closed 2022-01-05 by phoddie