Moddable-OpenSource/moddable #707: Application stop when touch screen before context of screen is set

stc1988, 2021-09-23:
**Build environment:**  Windows
**Target device:**M5stack core2

**Description**
Application stop when touch screen before context of screen is set

**Steps to Reproduce**
1. Launch `examples/piu/sound`.
2. Touch screen before screen shows.
3. See error `\moddable\build\devices\esp32\setup\piu.js (73) # Break: ?: cannot coerce undefined to object!`

**Images**

https://user-images.githubusercontent.com/11245747/134524821-e300988c-f71d-4b35-b4d7-c07bdc64c11d.MOV

**Other information**

As shown in the video,  `this.context` is not set.
I think it is better to check for `this.context` when screen touched.

phoddie, 2021-09-23:
Good find. There is a small window of time where that can happen. I tried fixing it in a little bit different way that you suggest. Let's see how that works out.

stc1988, 2021-09-24:
Thanks.
I confirmed this issue resolved.

closed 2021-09-24 by stc1988