Moddable-OpenSource/moddable #709: ESP32-C3 Support

diogoviannaaraujo, 2021-09-29:
Just got a ESP32-C3 dev board and was wondering if it was possible to use molddable with it. A simple mcconfig like a normal esp32 gives me a error:
`--chip esp32 failed`

ESP32-C3 is a nice update with lower costs, open architecture and has some nice crypto accelerators, if all it need is a new xsProj maybe I can try a pull request.

phoddie, 2021-09-29:
There's no support yet for for C3. There is support for the original ESP32, ESP32-S2, and ESP32-S3. Those may be good models for adding C3 support. A PR would be most welcome. ;)

LokiMetaSmith, 2022-02-08:
Luckly, it looks like the build tools are structured around esp-idf, which as of v4.4 works well with esp32c3. I believe, one could follow the instructions for https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md and update the latest version of esp-idf. The only tricky part is that we'll need to copy https://github.com/Moddable-OpenSource/moddable/tree/public/build/devices/esp32)/xsProj-esp32s3/ and make sure all of the compiler directives are correct. The esp32c3 uses a risc-v architecture, but that is handled by the esp-idf build system. I'll take a crack at this, as I need support for esp32c3 for my project.

phoddie, 2022-02-08:
@LokiMetaSmith  -- that would be **very** cool. Thank you! Feel free to reach out via GitHub discussions or our [Gitter](https://gitter.im/embedded-javascript/moddable) if we can help.

FWIW – we are currently working on ESP-IDF 4.4 support. It is mostly straightforward though has been at least one ugly surprise along the way...

phoddie, 2022-02-16:
@LokiMetaSmith – just FYI, the ESP-IDF 4.4 landed today. Hope that helps.

LokiMetaSmith, 2022-02-21:
I'm taking a stab at it on this branch. 
https://github.com/juniper-garden/moddable/tree/esp32c3_dev

Currently running into problems with 

# cc xsAll.c (strings in flash)
make: xtensa-esp32c3-elf-gcc: No such file or directory
make: *** [/home/loki/Projects/moddable/build/tmp/esp32/esp32c3/debug/helloworld/makefile:624: /xsAll.c.o] Error 127

I have seen a few other questions about this when I google "xtensa-esp32c3-elf-gcc", but they seem to be a one off error that isn't reproducible which makes me think it's probably a config variable in the build setup, but I'm not super familiar with how moddable modular build works. I've tried bypassing the build settings and use esp-idf idf.py directly without success.

phoddie, 2022-02-21:
> I've tried bypassing the build settings and use esp-idf idf.py directly without success.

Projects using the Moddable SDK won't build with the ESP-IDF alone because there are many tools in the Moddable SDK that are required to build the JavaScript firmware image.

> make: xtensa-esp32c3-elf-gcc: No such file or directory

This suggests that it cannot find the GCC tools for ESP32C3.

Try these to see which tools you have available:

> which xtensa-esp32-elf-gcc
> which xtensa-esp32-s2-elf-gcc
> which xtensa-esp32-s3-elf-gcc
> which xtensa-esp32c3-elf-gcc

I'm not sure if it is relevant but at `$HOME/.espressif/tools` (where xtensa-esp32-elf-gcc lives on my machine) there is also `riscv32-esp-elf`. Maybe that's the equivalent for the esp32c* targets?

mkellner, 2022-02-22:
Please try this patch:

```
diff --git a/tools/mcconfig/make.esp32.mk b/tools/mcconfig/make.esp32.mk
index 4db11da9..45e5cb52 100644
--- a/tools/mcconfig/make.esp32.mk
+++ b/tools/mcconfig/make.esp32.mk
@@ -29,7 +29,7 @@ EXPECTED_ESP_IDF ?= v4.4
 ESP32_SUBCLASS ?= esp32
 # $(warning ESP32_SUBCLASS $(ESP32_SUBCLASS))
 
-ifeq ($(ESP32_SUBCLASS),"esp32c3")
+ifeq ("$(ESP32_SUBCLASS)","esp32c3")
        ESP_ARCH = riscv
 else
        ESP_ARCH = xtensa
@@ -208,12 +208,22 @@ XS_HEADERS = \
        $(XS_DIR)/platforms/esp/xsPlatform.h
 HEADERS += $(XS_HEADERS)
 
+ifeq ("$(ESP32_SUBCLASS)","esp32c3")
+CC  = riscv32-esp-elf-gcc
+CPP = riscv32-esp-elf-g++
+LD  = $(CPP)
+AR  = riscv32-esp-elf-ar
+OBJCOPY = riscv32-esp-elf-objcopy
+OBJDUMP = riscv32-esp-elf-objdump
+else
 CC  = xtensa-$(ESP32_SUBCLASS)-elf-gcc
 CPP = xtensa-$(ESP32_SUBCLASS)-elf-g++
 LD  = $(CPP)
 AR  = xtensa-$(ESP32_SUBCLASS)-elf-ar
 OBJCOPY = xtensa-$(ESP32_SUBCLASS)-elf-objcopy
 OBJDUMP = xtensa-$(ESP32_SUBCLASS)-elf-objdump
+endif
+
 ESPTOOL = $(IDF_PATH)/components/esptool_py/esptool/esptool.py
 
 AR_FLAGS = crs
```

With this change, building for `-p esp32/esp32c3` finds the correct toolchain.

It doesn't get far though. Compiler flags need to be ported to the risc compiler.

```
$ mcconfig -d -m -p esp32/esp32c3
# cc xsAll.c (strings in flash)
riscv32-esp-elf-gcc: error: unrecognized command line option '-mlongcalls'
riscv32-esp-elf-gcc: error: unrecognized command line option '-mtext-section-literals'; did you mean '-fext-numeric-literals'?
make: *** [/Users/mkellner/github/esp32c3_dev/moddable/build/tmp/esp32/esp32c3/debug/lib/xsAll.c.o] Error 1
```

LokiMetaSmith, 2022-05-04:
It took me a while to come back to this, but I resolved that -mlongcalls and -mtext-section-literals are xtensa specific compiler commands for gcc, so I broke those out. I'm now getting a weird bug that seems to deal with how esp-idf defines it's header file namespace. The branch debug at https://github.com/juniper-garden/moddable/tree/debug has my current working stack and the above changes.

phoddie, 2022-06-30:
ESP32-C3 is now supported. Thank you to @LokiMetaSmith for his work to make this happen!

closed 2022-06-30 by phoddie