Moddable-OpenSource/moddable #712: esp32 Serial.onWritable is firing continuously

gavrilyak, 2021-10-02:
```
new device.io.Serial({
      baud: 9600,
      port: 1,
      format: "buffer",
      onWritable: function(cnt) {
         trace("writable", cnt, "\n")
      }
});
```

This code will rapidly trace "writable 128".

Proposed solution:
remove lines 377 and 378 in modules/io/serial/esp32/serial.c

P.S. 
And one more possible issue is in xs_serial_destructor. It calls   uart_enable_tx_intr with enable = 0, but in esp-idf code this parametr is ignored. Possibly it should be  uart_disable_tx_intr instead.

phoddie, 2021-10-02:
Thanks for the report.

The destructor issue is clearly a problem. It seems like a bug in the ESP IDF that `uart_enable_tx_intr` ignores the `enable` argument, but.... your solution is a good simplification.

The proposed fix to `onWritable` may be fine, but I want to run some tests before merging it.

gavrilyak, 2021-10-03:
Thank you.
Your tests are necessary because I have no idea whether this is a proper fix at all, I tried it and it seems to work for me, but this serial stuff is pretty complex (though you've made a fantastic job simplifying it) and it is also TC53 spec. If I was sure, I would have proposed a PR instead of issue.
It looks like it should be fine though, caller sets `onWritable` and calls `write`, which will call `uart_enable_tx_intr`, no need to enable interrupt after calling callback, "no write - no interrupt"  seems straightforward.
I am less sure about similar case in `onReadable`, I tried to remove  `uart_enable_rx_intr(serial->uart)` from `serialDeliver` and it works fine too, because a `read` call from user code will enable it anyway.

phoddie, 2021-10-04:
Your were on the right track when you removed the re-enabling of the interrupt from `serialDeliver` and relied on the enable in in `write`. That caused stalls with the `continuouswrite` example until I changed the reenable to only happen in `write` when it is disabled. I also cleaned up some redundancy -- `uart_enable_tx_intr` calls `uart_ll_set_txfifo_empty_thr` internally, so there's no need to call both.

That should be live in in the next day.

>  ..this serial stuff is pretty complex (though you've made a fantastic job simplifying it)

Thanks. This approach was the result of a failure of my first try, which failed completely.. It used the high level ESP-IDF UART API for reads and writes but it was incredibly complicated, never quite worked right, and used more memory.

phoddie, 2021-10-04:
Fix committed.

gavrilyak, 2021-10-05:
> It used the high level ESP-IDF UART API for reads and writes but it was incredibly complicated, never quite worked right, and used more memory.

I looked at ESP-IDF high level API source (interrupt handler in particular), so yeah, I can imagine the level of your struggle. But low-level seems to work just fine.

> Fix committed.

Thank you. I will verify and report back. Unforunately I do not have an access to hardware right now.

closed 2021-10-10 by phoddie