Moddable-OpenSource/moddable #713: SPI pins of esp32/Subplatform was changed

stc1988, 2021-10-03:
About https://github.com/Moddable-OpenSource/moddable/commit/93e6a19ba90559bff3f6f30dc45d9d3b079d60e1 ; 
 This is SPI pins  of `mc.defines.h`  in `esp32/m5stack` 

```
#define MODDEF_SPI_MOSI_PIN (13)
#define MODDEF_SPI_MISO_PIN (12)
#define MODDEF_SPI_SCK_PIN (14)
```

It's different from [manifest.json of m5stack](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack/manifest.json#L17-L21) . I think [defines of esp32](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/manifest.json#L22-L27) overwrites which subplatform defined.

phoddie, 2021-10-04:
Good catch! I accidentally broke that when implementing support for pin configuration to the Ecma-419 SPI implementation. We've already got a fix tested and will push that today.

phoddie, 2021-10-04:
The changes are committed.

stc1988, 2021-10-04:
I confirmed this issue fixed.

closed 2021-10-04 by stc1988