Moddable-OpenSource/moddable #722: Driver and Display Model for Moddable Two

markingle, 2021-10-09:
What driver and display model is used on the Moddable Two hardware?  I am trying to get familiar with the hardware using examples from the Bodmer eSPI library. Specifically I need to know the SPI pins to configure the header file.

If I can use the Moddable Two board for a project it will save me a ton of work.  Thanks in advance!

phoddie, 2021-10-09:
The [Moddable Two documentation ](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/moddable-two.md) and Moddable Two manifest file contain he answers to your questions.

- The display driver is connected to pins 13 (MOSI), 14 (CLK), 15 (CS), and 2 (DC).
- It can run at up to 40 MHz
- It uses the MIPI compatible ILI9341 display driver
- The [base registers](https://github.com/Moddable-OpenSource/moddable/blob/82db9e6d652ec474767b1772ac5d656d5b4960be/modules/drivers/ili9341/modIli9341.c#L474-L491) are in the ILI9341 driver and are augmented by [those](https://github.com/Moddable-OpenSource/moddable/blob/82db9e6d652ec474767b1772ac5d656d5b4960be/build/devices/esp32/targets/moddable_two/manifest.json#L34-L37) in the Moddable Two manifest

markingle, 2021-10-09:
Thanks Peter!  I have tried those pin settings in the Setup_User.h file with no luck.   But I will review the manifest to see if that helps....Thanks for the quick reply!

markingle, 2021-10-09:
Well I changed the SPI Frequency to 40Mhz and it started working.  I loaded a simple TFT_Print_Test  program using Arduino.  But for some reason after loading another program to test, the backlight (white) is on but nothing is showing on the screen.  I tried reloading the TFT_Print_Test again and now its not working either....weird! Any ideas?

phoddie, 2021-10-12:
The display controller in Moddable Two is very common. I'm certain it can work with the Adruino libraries. However, the Moddable doesn't have experience with that so we're limited in how we can help.

phoddie, 2021-10-12:
I'm moving this to discussions as it isn't an issue.