Moddable-OpenSource/moddable #731: storage size of 'tz' isn't known  ESP-IDF 8.3 and Windows

MKGaru, 2021-11-21:
**ModdableSDK:** public (ba9da8a9f6f)
**Build environment:** Windows 10/11
**Target device:** ESP32 (ESP-IDF release/8.3  98d34e5f6)

**Description**
A build error occurs when building a moddable and ESP32 environment on Windows.
It occurred either with an upgrade with a clean build of the existing environment or a setup to a new environment.

**Steps to Reproduce**
1. build hellowrold: `C:\Users\user\moddable\examples\helloworld>mcconfig -d -m -p esp32`
2.
C:\Users\user\moddable\xs\platforms\esp\xsHost.c: In function 'modSetTime':
C:\Users\user\moddable\xs\platforms\esp\xsHost.c:663:18: error: storage size of 'tz' isn't known

**Other information**
If comment out this line, can build and works.
https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c#L663

andycarle, 2021-11-21:
Hi @MKGaru, 

I am seeing some issues with the ESP-IDF commit that you mention (`98d34e5f6`). But, the build is working as expected with our recommended ESP-IDF commit: `v4.3.1`. 

Try:
```terminal
cd %IDF_PATH%
git checkout v4.3.1
git submodule update --init --recursive
```

Then clean your Moddable build output and try your build again.

MKGaru, 2021-11-21:
It's works! Thank you supporting.

closed 2021-11-21 by MKGaru