Moddable-OpenSource/moddable #732: storage size of 'tz' isn' known when building for moddable_two

ollex, 2021-11-26:
Hi, I just tried to go through the required setup steps and build the first example following [,](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md) all without any hiccups, but the final building process with mcconfig -d -m -p esp32/moddable_two  fails because:

/home/linux/ModdableProjects/moddable/xs/platforms/esp/xsHost.c: In function 'modSetTime':
/home/linux/ModdableProjects/moddable/xs/platforms/esp/xsHost.c:663:18: error: storage size of 'tz' isn't known
  struct timezone tz;
               
I don't have a board connected, but that shouldn't matter I guess as the moddable_two storage sizes should be known to the compiler/scripts ?

ollex, 2021-11-26:
I see something similar for windows in https://github.com/Moddable-OpenSource/moddable/issues/731 so will try to see if I have a different version of esp-idf than 4.31 (installed latest I guess following the instructions) and rebuild...

phoddie, 2021-11-26:
Please see issue #731. It reports the same issue and explains how to resolve it.

ollex, 2021-11-26:
Hm - after checking out v4.3.1 and running install.sh again it tells me 
# Reconfiguring ESP-IDF...
/bin/sh: 1: idf.py: not found

ollex, 2021-11-26:
I did run source export.sh again so not sure what's gone havoc...

andycarle, 2021-11-26:
That sounds like you probably did not re-run export.bat after running install.bat. I'd recommend opening up a new command prompt and running export.bat in it before proceeding. You might also want to clear out %MODDABLE%\build\bin\esp32 and %MODDABLE%\build\tmp\esp32.

ollex, 2021-11-26:
Hm I iwll try to clear out the build folder and a new terminal running export.sh again (did that but before but to no avail although it reported everything like before when it was ok)

ollex, 2021-11-26:
that helped - project build complete - great !

closed 2021-11-26 by ollex