Moddable-OpenSource/moddable #733: Build fails if first project build uses '-o' option

cmidgley, 2021-11-27:
**Build environment:** Verified on Windows and Linux
**Target device:** Various ESP32 targets (esp32, esp32/heltec_wifi_kit_32, esp32/heltec_lora_32)

**Description**
For a new project (one never built), attempts to build using `-o <directory>` will fail.

**Steps to Reproduce**
Run the following on a project that has never been built on the system:
```
mkdir compiled
mcconfig -m -p esp32 -t build -o compiled
### Error: No such file or directory
```

**Workaround**
If you do a build without `-o <directory>` once, it will then work.  It appears to be part of the `makefile` construction, as this will work on a new project:

```
mcconfig -p esp32 
mcconfig -m -p esp32 -t build -o compiled
```

**Other information**
I used `strace` in an attempt to unearth which file was the issue but learned nothing.  I ran a failure and success use case such as this:

`
strace -e trace=%file mcconfig -p esp32 -o compiled >test-1 2>&1
strace -e trace=%file mcconfig -p esp32 >test-2 2>&1
`

Comparing the `test-1` and `test-2` files show no difference other than the error message.

**Use case**
Using automated builds in a Docker container environment, where a fresh container is always a brand new build.

phoddie, 2021-11-30:
I was able to get this working on macOS. There were a two problems.

1. In mcmanifest.js, there are a couple places where it uses `tool.buildPath` where it should use `tool.outputPath`. Here are the two lines:
```
let idfBuildDir = tool.buildPath + "\\tmp\\" + tool.environment.PLATFORMPATH + "\\" + (tool.debug ? "debug\\idf" : "release\\idf");

let outputConfigDirectory = tool.buildPath + tool.slash + "tmp" + tool.slash + "esp32" + tool.slash + (tool.subplatform ?? "") + tool.slash + (tool.debug ? "debug" : (tool.instrument ? "instrument" : "release")) + tool.slash + tool.environment.NAME + tool.slash + "xsProj-";

```

2. In `tools/mcconfig/make.esp32.mk` all occurrences of `$(BUILD_DIR)/tmp` need to be replaced with `$(TMP_DIR)` For Windows, you'll need to make the corresponding changes to `nmake.esp32.mk`.

With those changes applied,  the build succeeds the first time. It completely avoids writing to $MODDABLE/build/*.  If that works for you too, I'll commit the changes.

For future debugging reference, I tracked this down by running the debug build of the mcconfig with xsbug active. That way it stops in the debugger on exceptions. To run the debug version, I redefine the path to Moddable tools in $PATH from `${MODDABLE}/build/bin/mac/release` to `${MODDABLE}/build/bin/mac/debug`.

cmidgley, 2021-11-30:
I tried this several times from a clean image, but it appears to break all builds on Linux.  

```
mcconfig -m -d -p esp32/heltec_lora_32 -o compiled -t build
python: can't open file '/root/Projects/moddable/examples/base/console/compiled/tmp/esp32/heltec_lora_32/debug/console/devices/esp32/xsProj-esp32/versionCheck.py': [Errno 2] No such file or directory
Expected ESP IDF v4.3.1, found v4.3.1
/root/Projects/moddable/examples/base/console/compiled/tmp/esp32/heltec_lora_32/debug/console/makefile:721: recipe for target 'idfVersionCheck' failed
make: *** [idfVersionCheck] Error 1
```

The same failure occurs independent of using `-o <directory`.

phoddie, 2021-11-30:
It is looking for`versionCheck.py` in your output directory (compiled). It should be looking in the build directory. Here's the line where `versionCheck.py` is invoked:

```
python $(PROJ_DIR_TEMPLATE)/versionCheck.py
```

In my generated makefile, there is:

```
BUILD_DIR = /Users/hoddie/Projects/moddable/build
TMP_DIR = /Users/hoddie/Projects/moddable/examples/helloworld/compile/tmp/esp32/release/helloworld
PROJ_DIR_TEMPLATE = $(BUILD_DIR)/devices/esp32/xsProj-$(ESP32_SUBCLASS)
```

It looks like your `BUILD_DIR` is pointing to `TMP_DIR`.

cmidgley, 2021-11-30:
RTFM!  It was an editing mistake (not TMP_DIR, but that hinted me into the mistake I made).  In `make.esp32.mk` I incorrectly changed all `$(BUILD_DIR)` to `$(TMP_DIR)` when you clearly had said only to replace `$(BUILD_DIR)/tmp`.  I did it again and it works great.  

FYI ... I was wondering if `mconfig ... -o <dir>` should create the directory if it doesn't exist?  It takes care of that when not using `-o`.  Right now I always do a `mkdir -p <dir> && mcconfig ...`, which works fine (and it's in a build script so it's automatic) but thought I'd mention it.

phoddie, 2021-11-30:
OK, good. I'll get that committed.

`mcconfig` probably should create output directory. But to do that would require another native function that works across platforms. That's more of an adventure than today can handle. ;)

phoddie, 2021-12-01:
The changes are now available for all platforms and targets. Please give it another try. My original fix was generating incorrect paths in when `-o` wasn't used. Thanks to @andycarle for fixing that.

closed 2021-12-01 by cmidgley