Moddable-OpenSource/moddable #736: Unable to catch errors in mod during initial load

cmidgley, 2021-12-03:
**Build environment:** Windows, Linux
**Target device:** ESP32

**Description**
Unable to catch runtime errors in a mod during `Modules.importNow(...)`.  Wrapping try/catch around `importNow` does not catch any exceptions.  When running, `xsbug` does a break into the debugger (see image below), and pressing continue reboots the ESP32.  

**Steps to Reproduce**
*Host file: main.js*
```
import Modules from "modules";

if (Modules.has("examplemod")) {
	try {
		trace("Importing mod\n");
		let exports = Modules.importNow("examplemod");
		trace("Import successful\n");
	} catch (err) {
		trace(`Caught error: ${err}\n`);
	}
} else trace("Mod 'examplemod' not available\n");
```
*Host file: manifest.json*
```
{
	"include": "$(MODDABLE)/examples/manifest_mod.json",
	"modules": {
		"*": ["./examplemod"]
	}
}
```
*Mod file: examplemod.js*
```
trace("Inside mod\n");
let a = b.c;  // error: variable b is undefined
trace("Mod global complete\n");
```
*Mod file: manifest.json*
```
{
	"include": "$(MODDABLE)/examples/manifest_mod.json",
	"modules": {
		"*": ["./examplemod"]
	}
}
```

**Expected behavior**
After loading the host (mcconfig) and the mod (mcrun) I would expect the trace output something like:

```
Importing mod
Inside mod
Caught error: get b: undefined variable!
```

**Actual behavior**
It breaks into the debugger (reporting the correct runtime error but it shouldn't breakpoint) and pressing continue causes the ESP32 to reboot with no further code execution.

[image]

**Other information**
If the code is modified to so that the mod doesn't have errors during load/setup, but exports logic that does have runtime errors, try/catch works correctly.  For example, a mod with:
```
export function demo() {
	let a = b.c;
}
```
will get correctly caught in the host with:
```
import Modules from "modules";

if (Modules.has("examplemod")) {
	let exports = Modules.importNow("examplemod");

	try {
		exports.demo();
		trace("Demo call worked\n");
	} catch (err) {
		trace(`Demo call failed: ${err}\n`);
	}
}
```

phoddie, 2021-12-04:
Thanks for writing this up. I tried on macOS and it worked correctly. I'll try on ESP32.

One detail - the host manifest about appears to be identical to the mod manifest. It may not be relevant to this issue, but it would be helpful to have the manifest you are using in case (for example, I assume your main.js is not preloaded).

cmidgley, 2021-12-04:
Thanks for looking into this so quickly!  Sorry about the host `manifest.json` file being a dup of the mod one - I've updated the original comment to have the correct file, which is:

```
{
        "include": ["$(MODDABLE)/examples/manifest_base.json", "$(MODULES)/base/modules/manifest.json"],
        "modules": {
                "*": ["./main"]
        },
        "defines": {
                "XS_MODS": 1
        }
}
```

phoddie, 2021-12-04:
Some news. I was able to reproduce this on ESP32 (but not macOS). It looks to be crashing in the exception case of the module loader. If you have core dumps enabled, you will see the crash in the console. Here's the relevant decoded output:

```
==================== CURRENT THREAD STACK =====================
#0  0x400e130e in fxRunImportNow (the=0x3ffb4aec, realm=0x3ffbf418, moduleID=0) at $MODDABLE/xs/sources/xsModule.c:1511
#1  0x400e1639 in fxAwaitImport (the=0x3ffb4aec, defaultFlag=1) at $MODDABLE/xs/sources/xsModule.c:1375
#2  0x400f6874 in xs_modules_importNow (the=0x3ffb4aec) at $MODDABLE/modules/base/modules/modules.c:75
#3  0x400e3ecd in fxRunID (the=0x3ffb4aec, generator=0x0, count=<optimized out>) at $MODDABLE/xs/sources/xsRun.c:891
#4  0x400e11c0 in fxRunImportNow (the=0x3ffb4aec, realm=0x3ffbf418, moduleID=<optimized out>) at $MODDABLE/xs/sources/xsModule.c:1484
#5  0x400e1639 in fxAwaitImport (the=0x3ffb4aec, defaultFlag=1) at $MODDABLE/xs/sources/xsModule.c:1375
#6  0x400f30d1 in modLoadModule (theIn=0x3ffb4aec, name=0x3f4008c4 \"main\") at $MODDABLE/xs/platforms/esp/xsHost.c:813
#7  0x400f321a in setStepDone (the=0x3ffb4aec) at $MODDABLE/xs/platforms/esp/xsHost.c:825
#8  0x400f38e8 in mc_setup (the=0x3ffb4aec) at $MODDABLE/xs/platforms/esp/xsHost.c:865
#9  0x400d6054 in loop_task (pvParameter=<optimized out>) at $MODDABLE/build/tmp/esp32/moddable_two/debug/issue726/xsProj-esp32/main/main.c:134
```

Not sure yet why macOS gets lucky and ESP32 doesn't: the code is the same.

phoddie, 2021-12-05:
Good news. We have a fix for this issue that tests successfully on ESP32, ESP8266, and macOS. It looks like it could have failed on all platforms, but I got lucky on macOS because of the values that happened to be in memory. We should push the fix in the next day or so.

Thanks for taking the time to write this one up. The issues as introduced with the recent revision of the module implementation.

closed 2021-12-21 by phoddie

cmidgley, 2022-01-14:
Finally can build my code again ... and I can confirm that this fix appears to be working great.  Thanks.