Moddable-OpenSource/moddable #746: AddressSanitizer: heap-buffer-overflow in fxRunID

eternalsakura, 2021-12-11:
- commit hash
db8f9735c20c07c189d7f4a0574e787d4a6669ea

- patch to enable mxNoChunks 
```
diff --git a/xs/sources/xsMemory.c b/xs/sources/xsMemory.c
index a8b888ff..b9df73a4 100644
--- a/xs/sources/xsMemory.c
+++ b/xs/sources/xsMemory.c
@@ -46,9 +46,9 @@
 #ifndef mxFill
 #define mxFill 0
 #endif
-#ifndef mxNoChunks
-#define mxNoChunks 0
-#endif
+// #ifndef mxNoChunks
+#define mxNoChunks 1
+// #endif
 
 #if mxStress
 int gxStress = 0;
@@ -223,6 +223,7 @@ void fxCheckCStack(txMachine* the)
     char x;
     char *stack = &x;
     if (stack <= the->stackLimit) {
+               exit(0);
        fxAbort(the, XS_STACK_OVERFLOW_EXIT);
     }
 }
```
- build with asan
```
make debug
```

- poc
```
const n = 130000;
{
    let x = new Set();
    for (let i = 0; i < n; ++i)
        x.add(i);
    let a = [...x];
}
{
    let x = new Set();
    for (let i = 0; i < n; ++i)
        x.add(i);
    let a = [...x.values()];
}
{
    let x = new Set();
    for (let i = 0; i < n; ++i)
        x.add(i);
    let a = [...x.keys()];
}
```

- release log
```
➜  moddable_workdir git:(main) ~/moddable/build/bin/lin/release/xst fuzz_output/fuzzer1/.cur_input
malloc(): invalid size (unsorted)
[1]    6727 abort      ~/moddable/build/bin/lin/release/xst fuzz_output/fuzzer1/.cur_input
...
Stopped reason: SIGABRT
__GI_raise (sig=sig@entry=0x6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
gdb-peda$ bt
#0  __GI_raise (sig=sig@entry=0x6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7c68859 in __GI_abort () at abort.c:79
#2  0x00007ffff7cd33ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7dfd285 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff7cdb47c in malloc_printerr (str=str@entry=0x7ffff7dffa50 "malloc(): invalid size (unsorted)") at malloc.c:5347
#4  0x00007ffff7cde234 in _int_malloc (av=av@entry=0x7ffff7e2eb80 <main_arena>, bytes=bytes@entry=0x9468) at malloc.c:3736
#5  0x00007ffff7ce02d4 in __GI___libc_malloc (bytes=0x9468) at malloc.c:3058
#6  0x0000555555601e6b in fxNewGrowableChunk ()
#7  0x0000555555614b95 in fxSetIndexProperty ()
#8  0x000055555561fd4c in fxRunID ()
#9  0x000055555562afd2 in fxRunScript ()
#10 0x0000555555675b35 in fxRunProgramFile ()
#11 0x0000555555583671 in main ()
#12 0x00007ffff7c6a0b3 in __libc_start_main (main=0x555555582e60 <main>, argc=0x2, argv=0x7fffffffdbf8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdbe8)
    at ../csu/libc-start.c:308
#13 0x0000555555583a1e in _start ()
```

- debug asan log
```
➜  moddable_workdir git:(main) ~/moddable/build/bin/lin/debug/xst fuzz_output/fuzzer1/.cur_input                                                
=================================================================
==5577==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60b000000628 at pc 0x0000004a62e0 bp 0x7ffe39128070 sp 0x7ffe39127838
WRITE of size 16 at 0x60b000000628 thread T0
    #0 0x4a62df in __asan_memcpy (/home/sakura/moddable/build/bin/lin/debug/xst+0x4a62df)
    #1 0x76fead in fxRunID /home/sakura/moddable/xs/sources/xsRun.c:2431:27
    #2 0x78d96f in fxRunScript /home/sakura/moddable/xs/sources/xsRun.c:4766:4
    #3 0x907654 in fxRunProgramFile /home/sakura/moddable/xs/tools/xst.c:1387:2
    #4 0x907654 in main /home/sakura/moddable/xs/tools/xst.c:281:8
    #5 0x7fecb680d0b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16
    #6 0x42e69d in _start (/home/sakura/moddable/build/bin/lin/debug/xst+0x42e69d)

0x60b000000628 is located 0 bytes to the right of 104-byte region [0x60b0000005c0,0x60b000000628)
allocated by thread T0 here:
    #0 0x4a6ddd in malloc (/home/sakura/moddable/build/bin/lin/debug/xst+0x4a6ddd)
    #1 0x6f7940 in fxFindChunk /home/sakura/moddable/xs/sources/xsMemory.c:390:10
    #2 0x6f7940 in fxNewGrowableChunk /home/sakura/moddable/xs/sources/xsMemory.c:1270:10

SUMMARY: AddressSanitizer: heap-buffer-overflow (/home/sakura/moddable/build/bin/lin/debug/xst+0x4a62df) in __asan_memcpy
Shadow bytes around the buggy address:
  0x0c167fff8070: fa fa fa fa fa fa fd fd fd fd fd fd fd fd fd fd
  0x0c167fff8080: fd fd fd fa fa fa fa fa fa fa fa fa fd fd fd fd
  0x0c167fff8090: fd fd fd fd fd fd fd fd fd fa fa fa fa fa fa fa
  0x0c167fff80a0: fa fa fd fd fd fd fd fd fd fd fd fd fd fd fd fa
  0x0c167fff80b0: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
=>0x0c167fff80c0: 00 00 00 00 00[fa]fa fa fa fa fa fa fa fa fa fa
  0x0c167fff80d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c167fff80e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c167fff80f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c167fff8100: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c167fff8110: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==5577==ABORTING
```

eternalsakura, 2021-12-11:
This is the first time I use this parameter to run fuzz, please check if this is a real problem

phoddie, 2021-12-11:
It looks like it may just be a problem in the `mxNoChunks`option. We're looking into it.

phoddie, 2021-12-13:
There is a problem with `mxNoChunks` with resizable chunks. We are testing a fix for that. Thank you.

phoddie, 2021-12-13:
(Confirmed as problem only in `mxNoChunks` test mode option)

closed 2021-12-13 by phoddie