Moddable-OpenSource/moddable #753: Cross platform breakpoints in xsbug fail work after device is reset or xsbug is restarted

cmidgley, 2021-12-15:
**Build environment:** Build firmware on Linux, run xsbug on Windows
**Target device:** ESP32

**Description**
When building firmware on Linux, and later debugging with `xsbug` on Windows, breakpoints and source code mapping works fine in the first debugging session.  However, when you restart the device the saved breakpoints fail to stop.  If you remove the breakpoints and add them again, it works until the next restart.

**Steps to Reproduce**
1. Build any example firmware under Linux, with ssid/password set on the build
2. Copy the bin directory to Windows
3. Flash the ESP32 device with the binary file
4. Start `xsbug` and `serial2xsbug`
5. Set the preferences feature to break on start
6. Reset the ESP32 and map to the Moddable `network.js` file locally
7. Set a couple breakpoints, such as on lines 30 (`if (!config.ssid) {`) and 35 (`let monitor = new WiFi...`)
8. Run the program, and see all breakpoints work
9. Reset the ESP32, which will break at start
10. Execute the code again, and notice no breakpoints work
11. Reset the ESP32 again
12. Clear and reset the breakpoints (clear all, or click on each, or just add new breakpoints)
13. Run and see the breakpoints work (new breakpoints work, old breakpoints are ignored)

Additionally, if you set the breakpoints, exit and restart `xsbug` the same behavior occurs.  This seems to imply that breakpoints are initially pathed correctly when setting them, but the act of persisting / reissuing them is somehow failing to use the same values.  

I checked the registry and the paths seem correct for the breakpoints, native to Windows, but I'm unclear if those paths should have been maintained in their original (Linux) form.  My attempts to follow how the source works have not been successful yet (I think `mxSetAllBreakpointsCommand` is involved in resetting these breakpoints, but my gut says the failure is earlier than that?)

phoddie, 2022-02-10:
We might have a fix. Would you give this a try? Replace this line:

https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/tools/xsbug/DebugBehavior.js#L715

with 

```js
items.push({ path, line, enabled:item.enabled });
```

and rebuild tools. Thank you.

cmidgley, 2022-02-10:
Super awesome!  Works great.  This will be a life saver.  THANKS!

phoddie, 2022-02-10:
Great. Thanks for confirming quickly. That'll be in our next update.

closed 2022-02-24 by phoddie