Moddable-OpenSource/moddable #756: ESP32 BLE Wifi connector service

radvansky-tomas, 2021-12-22:
I am trying to run basically any BLE Server example and connect to it existing proven mobile apps such as Lightblue, nRF Connect or BLE Hero just to try it out.

I have simple ESP32 dev board, I've installed all tooling and deployed example to board.

I can see and debug project.

However:

- I cannot connect to advertised BLE Device
- Therefore I cannot send any of required characteristics

AND for some reason
- I suspect that there is something going on with "caching", ie i can see different BLE Advertiser name than one in code etc...different number of services advertised etc...

Its like that whole wifi.json file is not being utilised.

radvansky-tomas, 2021-12-22:
And yes I just took another board (new sealed) and that works fine.

So my question is now more in realms of:

- How I can reset bluetooth stack (or whole esp32 for that matter).

I did erase firmware with esp tool, but this did nothing to bluetooth module

So I assume there should be way to directly reset that NRF soc or whatever that is

phoddie, 2021-12-22:
Erasing flash should return the board to its default state. There isn't a place for BLE to store dynamic state except flash:

```
esptool.py erase_flash
```

Is it possible that the board you are having trouble with is broken? You might check to see if Wi-Fi is working. The [Wi-Fi scan example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/wifi/wifiscan/main.js) is a good one to try.

radvansky-tomas, 2021-12-23:
I confirmed yesterday that there is no problem with the board/code itself. I could see correct BLE device name and characteristics when I tried to connect to it from different device. But for some reason apps/bt stack on my iphone still remembers old stuff...

andycarle, 2021-12-23:
For what it's worth, I've seen similar caching issues when working with BLE devices paired to iOS, Windows, and macOS.

To work around that, I've gotten in the habit of making small changes to the BLE advertising data as I go. Doing something minor like changing a character in the advertised Complete Name seems to be enough to convince the mobile/desktop OSes to recognize the ESP32 as a new device and not apply their cached GAP/GATT data.

And just to confirm what Peter noted above, my understanding is that the BLE pairing/bonding data on the ESP32 is stored in the NVS/preferences partition in flash. So doing a `esptool.py erase_flash` will clear that out.

phoddie, 2021-12-23:
> But for some reason apps/bt stack on my iphone still remembers old stuff...

The behavior of external BLE stacks (e.g. on a phone) is outside the scope of the Moddable SDK. Is there some indication of the ESP32 publishing incorrect or incomplete information that is confusing the phone? If not, this issue should be closed.

phoddie, 2021-12-28:
Closing this out as it appears to be a phone BLE cache issue that cannot be solved on the microcontroller.

closed 2021-12-28 by phoddie