Moddable-OpenSource/moddable #757: ESPIDF build failed while generating x509_crt_bundle

costa-victor, 2021-12-22:
## **Environment**
**Build environment:** Linux Mint 20 x86_64
**Target device:** NodeMCU ESP32
**Moddable version**: public - commit [e1043f5](https://github.com/Moddable-OpenSource/moddable/commit/e1043f55119a69bbe958e0e86ad418cfb4331e9e)
**ESPIDF version**:  [v4.3.1](https://github.com/espressif/esp-idf/tree/v4.3.1)

## **Description**
Compilation of helloworld example and others projects failed due to mbedtls in ESPIDF tag v4.3.1.

```bash
[84/86] Linking C static library esp-idf/main/libmain.a
[85/86] Linking C executable bootloader.elf
[86/86] Generating binary image from built executable
esptool.py v3.1-dev
Merged 1 ELF section
Generated /home/victor/moddable/build/tmp/esp32/nodemcu/debug/helloworld/xsProj-esp32/build/bootloader/bootloader.bin
[663/966] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cd /home/victor/moddable/build/tmp/esp32/nodemcu/debug/helloworld/xsProj-esp32/build/esp-idf/mbedtls && /home/victor/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/victor/esp/moddable/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/victor/esp/moddable/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q
gen_crt_bundle.py: Invalid certificate in /home/victor/esp/moddable/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
ESP-IDF Build Failed
make: *** [/home/victor/moddable/build/tmp/esp32/nodemcu/debug/helloworld/makefile:473: all] Erro 1
```

I started to investigate and this is a known error by ESPIDF that affects all of their MCU's, it's was detected in september/october after the **EC-ACC** certificate expires. It can be checked in details on issue [#7660](https://github.com/espressif/esp-idf/issues/7660) from ESPIDF.

## **Steps to Reproduce**
The easiest way to check this out:
1. In the **examples\hello** folder run `mcconfig -d -m -p esp32`
3. Shows the same mbedtls error above.

As expected, but tested to confirm:
* This also happens when i run without debug `mcconfig -m -p esp32`
* Or for differents targets based on ESP32, like nodemcu `mcconfig -d -m -p esp32/nodemcu`
* I haven't tested for ESP32-S2 and ESP32-S3, but this error may also appear.
* Compile ESPIDF projects using `idf.py build` gives the same error.

## **Expected behavior**
Complete the compilation process.

## **More information**
### ESPIDF Solution
I check the fix for this issue and it's done by removing the **EC-ACC** certificate from **esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem**, this is already done for **master**, **release/v4.2** and **release/v4.3** as shown [here](https://github.com/espressif/esp-idf/issues/7660#issuecomment-940917394).

However, this fix was applied after the tag v4.3.1, which is the ESPIDF version used as indicated on [step 4](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#installing-2).

* ESPIDF tag [v4.3.1](https://github.com/espressif/esp-idf/commit/2e74914051d14ec2290fc751a8486fb51d73a31e) - September 2nd.
* Fix for [release/v4.3](https://github.com/espressif/esp-idf/commit/caafeff63f0a3c7974e955c7af1e01006850074d) - October 7th.
* [Comparison](https://github.com/espressif/esp-idf/compare/v4.3.1...caafeff63f0a3c7974e955c7af1e01006850074d) between the commits above.

### Alternative solution
To avoid changing the currently tag used for moddable, it's possible bypass this certificate error as discussed in [#7621](https://github.com/espressif/esp-idf/issues/7621#issuecomment-931241064). It's might be a more safety solution rather then update the moddable dependency of ESPIDF to a new tag or release branch.

andycarle, 2021-12-28:
Thank you for the report and the suggested workaround!

Our preference is to correct this issue by bumping the supported ESP-IDF tag to v4.3.2. Doing so requires a small change to the Moddable SDK which will be included in an upcoming open source push. When that goes live, I'll link the relevant commit to this issue. 

&nbsp;- Andy

phoddie, 2021-12-29:
Closing as Moddable SDK bumped to ESP-IDF v4.3.2 eliminating the need for this change.

closed 2021-12-29 by phoddie