Moddable-OpenSource/moddable #762: Windows 10 - Building sample Hello World

js4iot, 2022-01-04:
**Build environment:**  Windows 10 (Virtual Machine)
**Target device:** I have a cheap [generic Wroom ESP32 board](https://www.amazon.ca/gp/product/B08YJCF6GF/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1), I am assuming it's an esp32/saola_wroom.

**Description**
Not able to follow the setup instructions to install and configure Moddable.

I created a new Windows 10 virtual machine to try to understand the installation. I got past the first part (Build and run applications for the desktop simulator) but now I am trying to do the second part (Part 2: Build and run applications on a development board).

I already have a working CP210x driver installed on the host machine, and using PuTTY on the virtual machine I can see the console.log statements from an application I built with Espruino. So I know the VM can see the ESP32.

I got these errors:

```
[90/90] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated C:/Users/andre/Projects/moddable/build/tmp/esp32/saola_wroom/debug/helloworld/xsProj-esp32s2/build/bootloader/bootloader.bin
[977/982] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
ccache C:\Users\andre\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\... **<TRUNCATED>** ain.dir/main.c.obj   -c ../main/main.c
../main/main.c: In function 'setup':
../main/main.c:144:35: warning: passing argument 4 of 'ESP_cloneMachine' makes integer from pointer without a cast [-Wint-conversion]
  gThe = ESP_cloneMachine(0, 0, 0, NULL);
                                   ^~~~
In file included from ../main/main.c:51:
C:/Users/andre/Projects/moddable/xs/platforms/esp/xsHost.h:255:20: note: expected 'uint32_t' {aka 'unsigned int'} but argument is of type 'void *'
  extern xsMachine *ESP_cloneMachine(uint32_t allocation, uint32_t stack, uint32_t slotCount, uint32_t keyCount, const char *name);
                    ^~~~~~~~~~~~~~~~
../main/main.c:144:9: error: too few arguments to function 'ESP_cloneMachine'
  gThe = ESP_cloneMachine(0, 0, 0, NULL);
         ^~~~~~~~~~~~~~~~
In file included from ../main/main.c:51:
C:/Users/andre/Projects/moddable/xs/platforms/esp/xsHost.h:255:20: note: declared here
  extern xsMachine *ESP_cloneMachine(uint32_t allocation, uint32_t stack, uint32_t slotCount, uint32_t keyCount, const char *name);
                    ^~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
The system cannot find the file specified.
The system cannot find the file specified.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.
```

**Steps to Reproduce**

These are the steps I followed for the [second part](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#win-instructions):

1. Downloaded and run the [ESP-IDF Tools Installer](https://dl.espressif.com/dl/esp-idf-tools-setup-2.4.exe). 
2. I Set environment variables IDF_PATH (%USERPROFILE%\OneDrive\Desktop\esp-idf) and PATH (%USERPROFILE%\.espressif\tools\idf-python\3.8.7, %USERPROFILE%\.espressif\tools\idf-python\3.8.7\Scripts at the start of the path)
3. Connected the ESP32 via USB
4. executed these commands, in order
```
"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"
cd %USERPROFILE%\OneDrive\Desktop\esp-idf
C:\Windows\system32\cmd.exe /k ""C:\Users\andre\.espressif\idf_cmd_init.bat" "C:\Users\andre\.espressif\python_env\idf4.3_py3.8_env\Scripts" "C:\Program Files\Git\cmd\""
cd %USERPROFILE%\Projects\moddable\examples\helloworld
%USERPROFILE%\.espressif\tools\idf-python\3.8.7\python.exe -m pip install --upgrade pip
python -m pip install -r %IDF_PATH%\requirements.txt
cls
mcconfig -d -m -p esp32/saola_wroom
```

**Expected behavior**
I am able to install and configure Moddable SDK and the code in the ESP32 board.
[... 345 lines cut ...]
  extern xsMachine *ESP_cloneMachine(uint32_t allocation, uint32_t stack, uint32_t slotCount, uint32_t keyCount, const char *name);
                    ^~~~~~~~~~~~~~~~
../main/main.c:144:9: error: too few arguments to function 'ESP_cloneMachine'
  gThe = ESP_cloneMachine(0, 0, 0, NULL);
         ^~~~~~~~~~~~~~~~
In file included from ../main/main.c:51:
C:/Users/andre/Projects/moddable/xs/platforms/esp/xsHost.h:255:20: note: declared here
  extern xsMachine *ESP_cloneMachine(uint32_t allocation, uint32_t stack, uint32_t slotCount, uint32_t keyCount, const char *name);
                    ^~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
The system cannot find the file specified.
The system cannot find the file specified.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.

C:\Users\andre\Projects\moddable\examples\helloworld>
C:\Users\andre\Projects\moddable\examples\helloworld>
```

andycarle, 2022-01-04:
Hi @js4iot, I'm sorry to hear that you're having trouble building. Let's see if we can get it sorted out.

> Target device: I have a cheap generic Wroom ESP32 board, I am assuming it's an esp32/saola_wroom.

[Saola](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html) is an ESP32-S2 based development board. The generic Wroom you linked to is based on the ESP32, not the ESP32-S2 (which is a different MCU).

I'd suggest trying again with just the `esp32` platform. So, your build command would be:

```text
mcconfig -d -m -p esp32
```

Please give that a shot and let me know how it goes with your dev board.

&nbsp;- Andy

js4iot, 2022-01-04:
Thanks, I tried just "esp32" and I got a bit further but still got some errors. This is the bottom of the output:

```
[969/969] Generating binary image from built executable
esptool.py v3.2-dev
Merged 2 ELF sections
Generated C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/build/xs_esp32.bin
Executing action: flash
Serial port COM3
Connecting.......................
COM3 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.
```

I am running a Windows 10 virtual machine inside a Windows 10 computer because I want to test the install scripts before I make modifications to my original computer.

I do have a device in COM3, I can actually communicate to it (from inside the VM) using PuTTY and see the console.log() statements generated by an existing Espruino application I have on that board.

js4iot, 2022-01-04:
I tried executing this statement `set UPLOAD_PORT=COM3` before `mcconfig -d -m -p esp32` and I go slightly different output but basically the same error:

```
-- Build files have been written to: C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/build/bootloader
ninja: no work to do.
Executing action: flash
Running ninja in directory c:\users\andre\projects\moddable\build\tmp\esp32\debug\helloworld\xsproj-esp32\build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\andre\OneDrive\Desktop\esp-idf\components\esptool_py && C:\Users...32/build" -P C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 xs_esp32.bin
esptool.py v3.2-dev
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
CMake Error at run_serial_tool.cmake:50 (message):
  C:/Users/andre/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe
  C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D C:\Users\andre\OneDrive\Desktop\esp-idf\components\esptool_py && C:\Users\andre\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/Users/andre/OneDrive/Desktop/esp-idf" -D SERIAL_TOOL="C:/Users/andre/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/build" -P C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.
```

js4iot, 2022-01-04:
OK, so I tried something else... as soon as I see the `Connecting............` message, I pressed and released the `BOOT` button and I got a different error (maybe better). This is what I am getting now:

```
Serial port COM3
Connecting............
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: c8:c9:a3:f9:a6:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Failed to set baud rate 921600. The driver may not support this rate.
CMake Error at run_serial_tool.cmake:50 (message):
  C:/Users/andre/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe
  C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D C:\Users\andre\OneDrive\Desktop\esp-idf\components\esptool_py && C:\Users\andre\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/Users/andre/OneDrive/Desktop/esp-idf" -D SERIAL_TOOL="C:/Users/andre/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/build" -P C:/Users/andre/OneDrive/Desktop/esp-idf/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.
```

js4iot, 2022-01-04:
OK, so I tried adding this line `set UPLOAD_SPEED=115200` and I got a bit farther, but still no luck...

```
Connecting...................
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:c9:a3:f9:a6:f0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00064fff...
Compressed 3072 bytes to 123...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (123 compressed) at 0x00008000 in 0.1 seconds (effective 347.7 kbit/s)...
Hash of data verified.
Compressed 17632 bytes to 12211...
Writing at 0x00001000... (100 %)
Wrote 17632 bytes (12211 compressed) at 0x00001000 in 2.1 seconds (effective 66.8 kbit/s)...
Hash of data verified.
Compressed 346496 bytes to 200156...
Writing at 0x00010000... (7 %)
Writing at 0x0001ae1a... (15 %)
Writing at 0x000252a3... (23 %)
Writing at 0x0002a691... (30 %)
Writing at 0x00030d96... (38 %)
Writing at 0x00035f6c... (46 %)
Writing at 0x0003b74e... (53 %)
Writing at 0x000412ea... (61 %)
Writing at 0x00048510... (69 %)
Writing at 0x0004d6c9... (76 %)
Writing at 0x00052abd... (84 %)
Writing at 0x00058dac... (92 %)
Writing at 0x00063350... (100 %)
Wrote 346496 bytes (200156 compressed) at 0x00010000 in 39.3 seconds (effective 70.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.
```

At this moment, I do not have any useful information as to what the error could be, so I am giving up. Any help, will be appreciated.

Frida854, 2022-01-04:
What about to erase the chip, it sometimes help, if there have been another program on it.
try:
```
esptool.py --port [/dev/ttyUSB0|COM1] --baud 115200 erase_flash
```

js4iot, 2022-01-04:
I will try this. 

Question. If I erase the chip, I guess it's like `format c:`, how do I get the OS back? Or whatever raw files it needs? Will the mcconfig mount that?

Frida854, 2022-01-04:
Yes after erase, you can use mcconfig again.

js4iot, 2022-01-04:
OK, so I did `esptool.py --port %UPLOAD_PORT% --baud=%UPLOAD_SPEED% erase_flash` and then `mcconfig -d -m -p esp32`. This is the output, I think the same output as this morning.

```
Connecting...........................
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:c9:a3:f9:a6:f0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00064fff...
Compressed 3072 bytes to 123...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (123 compressed) at 0x00008000 in 0.0 seconds (effective 934.9 kbit/s)...
Hash of data verified.
Compressed 17632 bytes to 12211...
Writing at 0x00001000... (100 %)
Wrote 17632 bytes (12211 compressed) at 0x00001000 in 1.0 seconds (effective 146.6 kbit/s)...
Hash of data verified.
Compressed 346496 bytes to 200156...
Writing at 0x00010000... (7 %)
Writing at 0x0001ae1a... (15 %)
Writing at 0x000252a3... (23 %)
Writing at 0x0002a691... (30 %)
Writing at 0x00030d96... (38 %)
Writing at 0x00035f6c... (46 %)
Writing at 0x0003b74e... (53 %)
Writing at 0x000412ea... (61 %)
Writing at 0x00048510... (69 %)
Writing at 0x0004d6c9... (76 %)
Writing at 0x00052abd... (84 %)
Writing at 0x00058dac... (92 %)
Writing at 0x00063350... (100 %)
Wrote 346496 bytes (200156 compressed) at 0x00010000 in 14.4 seconds (effective 193.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
C:\Users\andre\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(142): The parameter is incorrect.
NMAKE : fatal error U1077: 'C:\Users\andre\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.
```

js4iot, 2022-01-04:
Unfortunately, it's not telling me which parameter is incorrect so I do not have much to chew on :-)

Frida854, 2022-01-04:
I have a esp32-pico-kit and it is working and a moddable_two.
Have you tried to run xsbug in a cmd prompt? Does it open?
I am running on a new windows 11.

andycarle, 2022-01-04:
The good news is that it appears that your build is working correctly now and that your app is successfully being flashed to the ESP32. The failure appears to be with the debugging connection—It looks like there's a problem opening that connection to your dev board at the specified baud rate. 

Try setting one more environment variable:

```text
set DEBUGGER_SPEED=115200
```

That will configure the debugger to run at 115200, the same as you're successfully programming the device at.

You may have to do a clean build to get this to work correctly. The most thorough clean would be to simply delete the `%MODDABLE%\build\bin\esp32` and `%MODDABLE%\build\tmp\esp32` directories, after closing xsbug.

js4iot, 2022-01-04:
@Frida854 I opened xsbug from the command prompt and it opened a GUI application (see screenshot). I do not have a way to start it because the options are grayed out

[image]
.

js4iot, 2022-01-04:
I had a good feeling because I was making progress, but now I think that I trashed my ESP32 board. Now it goes into an infinite loop...

```
FC: cannot open C:\USERS\ANDRE\PROJECTS\MODDABLE\BUILD\TMP\ESP32\DEBUG\HELLOWORLD\XSPROJ-ESP32\PARTITIONS.CSV - No such file or folder

        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory c:\users\andre\projects\moddable\build\tmp\esp32\debug\helloworld\xsproj-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=C:/Users/andre/Projects/moddable/build/tmp/esp32//debug/helloworld/xsProj-esp32/sdkconfig.mc -DSDKCONFIG_HEADER=C:\Users\andre\Projects\moddable\build\tmp\esp32\debug\helloworld\xsProj-esp32\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=115200 -DIDF_TARGET=esp32 -DESP32_SUBCLASS=esp32 -DSDKCONFIG_DEFAULTS=C:\Users\andre\Projects\moddable\build\tmp\esp32\\debug\helloworld\xsProj-esp32\sdkconfig.mc -DCCACHE_ENABLE=1 c:\users\andre\projects\moddable\build\tmp\esp32\debug\helloworld\xsproj-esp32"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/andre/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/andre/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/andre/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/andre/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/andre/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/sdkconfig
Loading defaults file C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/sdkconfig.mc...
C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/sdkconfig.mc:230 CONFIG_ESP_CONSOLE_UART_NONE was replaced with CONFIG_ESP_CONSOLE_NONE
-- Found PythonInterp: C:/Users/andre/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe (found version "3.8.7")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "xs_esp32" version: OS210826-407-g83dadd3d
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/andre/Projects/moddable/build/tmp/esp32/debug/helloworld/xsProj-esp32/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/andre/OneDrive/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/andre/OneDrive/Desktop/esp-idf/components/app_trace C:/Users/andre/OneDrive/Desktop/esp-idf/components/app_update C:/Users/andre/OneDrive/Desktop/esp-idf/components/asio C:/Users/andre/OneDrive/Desktop/esp-idf/components/bootloader C:/Users/andre/OneDrive/Desktop/esp-idf/components/bootloader_support C:/Users/andre/OneDrive/Desktop/esp-idf/components/bt C:/Users/andre/OneDrive/Desktop/esp-idf/components/cbor [... cut ...]
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
NMA^KCE : fatTerminate batch job (Y/N)? al error U1058: terminated by user
Stop.
```

andycarle, 2022-01-04:
I'll think we'll be able to get you and your board recovered from this; I think it's just trying to boot a fully erased flash at the moment.

The build error you're getting now is one I'm familiar with: you're hitting a Windows path length limit for CCACHE in the ESP-IDF build. You can work around that by creating an output directory with a short path name (I recommend something simple like `C:\out\` and then building to it using the `-o` flag for `mcconfig`. 

1) Do something like this once to make the output directory:

```text
cd C:\
mkdir out
```

2) Then when you build, do:

```text
mcconfig -d -m -p esp32 -o c:\out
```

This is a real and annoying issue that comes up a lot with ESP-IDF builds even outside the Moddable ecosystem. But I would love to find a more automatic workaround for our developers...

js4iot, 2022-01-05:
I am building a course for developers that I am calling "Introduction to IoT with microcontrollers using JavaScript" so I have researched many different platforms like Espruino, Moddable, Mongoose, Low.js, J5e, etc.

One of those tools asks you to build the code in `C:\ESP32` folder, but moddable asks you to build it in "C:\Users\andres\Projects", that is definitely a lot more characters. Maybe it could be as simple as changing the instructions where this should be built :-)

andycarle, 2022-01-05:
Well, I appreciate your working through the issues here with us. We continuously update our Getting Started guides based on feedback we receive here, on Gitter, etc.

The majority of our developers do their work on macOS, so the Windows instructions and procedure have historically lagged a bit behind. I'm *very* open to suggestions on how to improve the process on Windows and would be delighted to work with you on implementing any ideas you have.

js4iot, 2022-01-05:
@andycarle I am taking good notes of the process to install this in Windows and I will love to help out improving the instructions provided. I have a few ideas that I could share later.

I am even thinking about reviewing the work done by @HipsterBrown using ZX [(https://github.com/HipsterBrown/xs-setup)](https://github.com/HipsterBrown/xs-setup) but adapting it to Windows. I fist must understand the process and get this darn Hello World working :-)

js4iot, 2022-01-05:
@andycarle earlier you asked me to erase the flash from ESP32 using this `esptool.py --port COM3 --baud 115200 erase_flash`. That may have been working out but it froze my computer (the virtual machine) and I have to kill it before I can move any further. 

Now that I am redirecting to C:\ESP32.out seems to be doing something similar... it gets this far and then the VM crashes. It stops responding and I have to shut it down ("unplug it") and restart it.

[image]

[image]

HipsterBrown, 2022-01-05:
@js4iot I'd like to include Windows support in `xs-setup`, so please reach out when you're working on automating it. There's some planned refactoring for that tool to make it more platform-agnostic and maintainable now that the concept has been prototyped using `zx`. 

I'm guessing this discussion might be why @phoddie reached out about prebuilding Moddable binaries 😄

js4iot, 2022-01-06:
Let me try to get this installed first on Windows manually. So far I have had nothing but problems ;-(

phoddie, 2022-01-07:
Moving to discussion, as this isn't a report of a specific issue.