Moddable-OpenSource/moddable #765: Stack-overflow xs/sources/xsMemory.c:986:22 in fxMarkValue

hope-fly, 2022-01-07:
##### Moddable-XS revision

Commit: [2f93df29](https://github.com/Moddable-OpenSource/moddable/commit/2f93df297782cf0a31c1797b246becbf5e546f4f)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
#debug
make -f xst.mk
```

##### Test case

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
function JSEtest() {
    var methods = 'map';
    var get;
    var p = new (Int16Array)('0123456789');
    for (var i = 0; i < methods.length; i += 1) {
        get = [];
        new (Int16Array)(4)['reduceRight'].call(p, Array);
        if ('') {
            return false;
        }
    }
    return true;
}

JSEtest();
```

</code></pre>
</details>

##### Execution  & Output

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js
AddressSanitizer:DEADLYSIGNAL
=================================================================
==118133==ERROR: AddressSanitizer: stack-overflow on address 0x7fff760d5c68 (pc 0x00000075c7c8 bp 0x7fff760d6190 sp 0x7fff760d5c70 T0)
    #0 0x75c7c7 in fxMarkValue /root/moddable/xs/sources/xsMemory.c:986:22
    #1 0x76b376 in fxMarkInstance /root/moddable/xs/sources/xsMemory.c:709:6
    #2 0x75c326 in fxMarkValue /root/moddable/xs/sources/xsMemory.c:965:4
    #3 0x75c7cc in fxMarkValue /root/moddable/xs/sources/xsMemory.c:986:5
    #4 0x76b376 in fxMarkInstance /root/moddable/xs/sources/xsMemory.c:709:6
    #5 0x75c326 in fxMarkValue /root/moddable/xs/sources/xsMemory.c:965:4
    #6 0x75c7cc in fxMarkValue /root/moddable/xs/sources/xsMemory.c:986:5
	......						......						......
	......						......						......
    #246 0x75c7cc in fxMarkValue /root/moddable/xs/sources/xsMemory.c:986:5
    #247 0x76b376 in fxMarkInstance /root/moddable/xs/sources/xsMemory.c:709:6
    #248 0x75c326 in fxMarkValue /root/moddable/xs/sources/xsMemory.c:965:4
    #249 0x75c7cc in fxMarkValue /root/moddable/xs/sources/xsMemory.c:986:5

SUMMARY: AddressSanitizer: stack-overflow /root/moddable/xs/sources/xsMemory.c:986:22 in fxMarkValue
==118133==ABORTING

```

phoddie, 2022-01-07:
This is a known issue with the garbage collector. For fuzz testing purposes, at least, we should invoke the stack check. The  Hardening XS article mentions this under [Vulnerability Testing](https://www.moddable.com/hardening-xs#advice-vulnerability):

> Native stack overflows are one potential parser vulnerability which XS only partially defends against. Issue reports here are still expected.

Issue #747 is another example of this.

phoddie, 2022-02-01:
The garbage collector now handles this case correctly.

closed 2022-02-09 by phoddie