Moddable-OpenSource/moddable #766: SEGV xs/sources/xsArray.c:2237:7 in fx_Array_prototype_sort

hope-fly, 2022-01-07:
##### Moddable-XS revision

Commit: [2f93df29](https://github.com/Moddable-OpenSource/moddable/commit/2f93df297782cf0a31c1797b246becbf5e546f4f)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
#debug
make -f xst.mk
```

##### Test case

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
var size = 256;
var array1 = new Array(size);

function toStr() {
  array1.splice(0, 2);
  return array1.sort().toString();
}

function JSEtest() {
  for (var i = 0; i < size; i++) {
    array1[i] = new Array(i);
    array1.sort()[i].toString = toStr;
  }
  array1.sort();
}

JSEtest();
```

</code></pre>
</details>

##### Execution  & Output

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js

AddressSanitizer:DEADLYSIGNAL
=================================================================
==101668==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000020 (pc 0x00000057e1b4 bp 0x7ffe5dc36870 sp 0x7ffe5dc35840 T0)
==101668==The signal is caused by a READ memory access.
==101668==Hint: address points to the zero page.
    #0 0x57e1b3 in fx_Array_prototype_sort /root/moddable/xs/sources/xsArray.c:2237:7
    #1 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #2 0x8ceaac in fxRunScript /root/moddable/xs/sources/xsRun.c:4766:4
    #3 0xad3231 in fxRunProgramFile /root/moddable/xs/tools/xst.c:1387:2
    #4 0xacfa83 in main /root/moddable/xs/tools/xst.c:281:8
    #5 0x7faf269ccbf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #6 0x42ddc9 in _start (/usr/local/bin/xst+0x42ddc9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/moddable/xs/sources/xsArray.c:2237:7 in fx_Array_prototype_sort
==101668==ABORTING
```

hope-fly, 2022-01-13:
When the testcase changed a little, there's a new crash during fuzzing. I'm not sure whether it's a duplicated issue or not.
If its not, I'll open a new issue. Hope the info listed as follows will be helpful!

##### Test case

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
var size = 1000;
var array1 = new Array(size);

function toStr() {
    array1.splice(0, 2);
    return array1.toString();
}

function JSEtest() {
    for (var i = 0; i < size; i++) {
        array1[i] = new Array(i);
        array1.sort()[i].toString = toStr;
    }
    array1.sort();
}

JSEtest();
```

</code></pre>
</details>

##### Execution  & Output

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js

AddressSanitizer:DEADLYSIGNAL
=================================================================
==39035==ERROR: AddressSanitizer: SEGV on unknown address 0x0000000004c0 (pc 0x00000059c0a6 bp 0x7ffc9a80fa50 sp 0x7ffc9a80f730 T0)
==39035==The signal is caused by a READ memory access.
==39035==Hint: address points to the zero page.
    #0 0x59c0a5 in fxCompareArrayItem /home/f1yfuzz/moddable/xs/sources/xsArray.c:350:18
    #1 0x57dc9a in fx_Array_prototype_sort /home/f1yfuzz/moddable/xs/sources/xsArray.c:2235:14
    #2 0x84f3ca in fxRunID /home/f1yfuzz/moddable/xs/sources/xsRun.c:842:7
    #3 0x8ceaac in fxRunScript /home/f1yfuzz/moddable/xs/sources/xsRun.c:4766:4
    #4 0xad3231 in fxRunProgramFile /home/f1yfuzz/moddable/xs/tools/xst.c:1387:2
    #5 0xacfa83 in main /home/f1yfuzz/moddable/xs/tools/xst.c:281:8
    #6 0x7f8d44116bf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #7 0x42ddc9 in _start (/usr/local/bin/xst+0x42ddc9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/f1yfuzz/moddable/xs/sources/xsArray.c:350:18 in fxCompareArrayItem
==39035==ABORTING
```

phoddie, 2022-01-13:
Thanks! From a quick look, it is tough to say if these are the same cause or not – the tests and crash are quite similar but the crash is in a different place. We'll investigate and let you know.

phoddie, 2022-02-01:
This has been resolved in our most recent Moddable SDK update. There were changes to Array sorting and a change to xst to ensure a clean exit when the JavaScript stack overflow is detected. Thank you again for the report.

closed 2022-02-09 by phoddie