Moddable-OpenSource/moddable #767: SEGV xs/sources/xsFunction.c:546:18 in fx_Function_prototype_hasInstance

hope-fly, 2022-01-07:
##### Moddable-XS revision

Commit: [2f93df29](https://github.com/Moddable-OpenSource/moddable/commit/2f93df297782cf0a31c1797b246becbf5e546f4f)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
make -f xst.mk
```

##### Test case

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
function JSEtest() { };

JSEtest.prototype = Function.prototype;
var obj = new JSEtest;

try {
    obj.apply();
    $ERROR('###');
} catch (e) {
    if (!(e instanceof new JSEtest)) {
        $ERROR('###');
    }
}
```

</code></pre>
</details>

##### Execution  & Output

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js

AddressSanitizer:DEADLYSIGNAL
=================================================================
==16775==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x00000069f0da bp 0x7ffc5bbdaf10 sp 0x7ffc5bbdac70 T0)
==16775==The signal is caused by a READ memory access.
==16775==Hint: address points to the zero page.
    #0 0x69f0d9 in fx_Function_prototype_hasInstance /root/moddable/xs/sources/xsFunction.c:546:18
    #1 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #2 0x8c98c0 in fxRunInstanceOf /root/moddable/xs/sources/xsRun.c:4396:2
    #3 0x8bac67 in fxRunID /root/moddable/xs/sources/xsRun.c:3877:4
    #4 0x8ceaac in fxRunScript /root/moddable/xs/sources/xsRun.c:4766:4
    #5 0xad3231 in fxRunProgramFile /root/moddable/xs/tools/xst.c:1387:2
    #6 0xacfa83 in main /root/moddable/xs/tools/xst.c:281:8
    #7 0x7f729fce6bf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #8 0x42ddc9 in _start (/usr/local/bin/xst+0x42ddc9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/moddable/xs/sources/xsFunction.c:546:18 in fx_Function_prototype_hasInstance
==16775==ABORTING
```

phoddie, 2022-01-07:
This looks like a duplicate of #748.

hope-fly, 2022-01-08:
Okay. I'll check the issues reported carefully before the next submitting.

closed 2022-01-10 by phoddie