Moddable-OpenSource/moddable #769: Negative-size-param (/usr/local/bin/xst+0x4ed5ec) in __asan_memmove

hope-fly, 2022-01-07:
##### Moddable-XS revision

Commit: [2f93df29](https://github.com/Moddable-OpenSource/moddable/commit/2f93df297782cf0a31c1797b246becbf5e546f4f)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
make -f xst.mk
```

##### Test case

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
var arr = [];
for (var i = 0; i < 28000; i++) {
    arr.push(new RegExp("ACAAAAATTAGCCGGGCGTGGTGGCGCGCGCCTGTAATCCCA" + i.toString(3)));
}
```

</code></pre>
</details>

##### Execution  & Output

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js
=================================================================
==106165==ERROR: AddressSanitizer: negative-size-param: (size=-2147483584)
    #0 0x4ed5ec in __asan_memmove (/usr/local/bin/xst+0x4ed5ec)
    #1 0x762e01 in fxSweep /root/moddable/xs/sources/xsMemory.c:1629:6
    #2 0x75a711 in fxCollect /root/moddable/xs/sources/xsMemory.c:278:3
    #3 0x767e87 in fxFindChunk /root/moddable/xs/sources/xsMemory.c:407:3
    #4 0x7678dd in fxNewChunk /root/moddable/xs/sources/xsMemory.c:1256:10
    #5 0x9a70e4 in fxCompileRegExp /root/moddable/xs/sources/xsre.c:1697:13
    #6 0x8469f7 in fxInitializeRegExp /root/moddable/xs/sources/xsRegExp.c:138:7
    #7 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #8 0x845fb6 in fx_RegExp /root/moddable/xs/sources/xsRegExp.c:235:2
    #9 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #10 0x8ceaac in fxRunScript /root/moddable/xs/sources/xsRun.c:4766:4
    #11 0xad3231 in fxRunProgramFile /root/moddable/xs/tools/xst.c:1387:2
    #12 0xacfa83 in main /root/moddable/xs/tools/xst.c:281:8
    #13 0x7f80e3cdfbf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #14 0x42ddc9 in _start (/usr/local/bin/xst+0x42ddc9)

0x7f80e004d7f0 is located 331760 bytes inside of 16777248-byte region [0x7f80dfffc800,0x7f80e0ffc820)
allocated by thread T0 here:
    #0 0x4edc80 in malloc (/usr/local/bin/xst+0x4edc80)
    #1 0x7dba63 in fxAllocateChunks /root/moddable/xs/sources/xsPlatforms.c:123:9
    #2 0x759641 in fxGrowChunks /root/moddable/xs/sources/xsMemory.c:506:11
    #3 0x75876a in fxAllocate /root/moddable/xs/sources/xsMemory.c:170:2
    #4 0x53d89c in fxCreateMachine /root/moddable/xs/sources/xsAPI.c:1382:4
    #5 0xace769 in main /root/moddable/xs/tools/xst.c:259:19
    #6 0x7f80e3cdfbf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310

SUMMARY: AddressSanitizer: negative-size-param (/usr/local/bin/xst+0x4ed5ec) in __asan_memmove
==106165==ABORTING
```

#### No-ASAN Output

```bash
[1]    131060 segmentation fault  xst poc.js
```

phoddie, 2022-01-10:
Interestingly, this issue has nothing to do with RegExp, but an unfortunate rounding case in converting an integer to string in base 3 (tracked down by @patrick-soquet). Fix pending.

hope-fly, 2022-01-13:
Verified!

closed 2022-01-13 by phoddie