Moddable-OpenSource/moddable #772: Provide additional control over typescript compilation options

cmidgley, 2022-01-09:
**Feature request**

Ability to have more control over the use of TypeScript from a `manifest.json` file.  I have two specific needs - to change the name of the compiler from `tsc` and to add additional fields to `tsconfig.json` (similar to the `types` feature).  A possible syntax to consider:

```json
{
    "compiler": "npx ttsc",
    "tsconfig": {
        "types": [
            "path/to/my/types"
        ],
       "plugins": [
           { "transform": "path/to/my/plugin" }
       ],
      "any-other-options": "etc."
    }
}
```

The above would run `npx ttsc` instead of `tsc` as the compiler, and would add the extra fields specified to `tsconfig.json` similar to how the new `types` field works.

> Note that I used the generic key `compiler` above, as that is the key used by many other tools in the eco-system for TypeScript compilation (json, ts-node, etc).  However, for Moddable, it may make more sense to name it `typescript-compiler` or `typescriptCompiler`, leaving open the future ability to change other compilers (such as front-ending `xsc`).

**Why do you think this feature would be useful?**

Since `tsconfig.json` is auto-generated, it leaves little control to the developer to use advanced TypeScript features.  Currently you cannot add options to `tsconfig.json` other than the few `mcconfig` allows, and you cannot inject different parameters or change the compiler used for typescript itself.  This proposal would open-end many options for controlling typescript execution without requiring hacking the `mcconfig` source.

**Use case**

I need to use [ttypescript](https://github.com/cevek/ttypescript) which allows me to inject a transformer script into TypeScript to expand it's generation capabilities.  This requires I run `ttsc` (or in my case, since it is not installed globally, `npx ttsc`).  Additionally it needs additional fields inside `tsconfig.json` to specify the path to the plugin(s) it should use.

The reason behind this is to use [SOLID design principles](https://en.wikipedia.org/wiki/SOLID) for firmware development, specifically the [Dependency Inversion Principle](https://en.wikipedia.org/wiki/Dependency_inversion_principle) via [Dependency Injection](https://en.wikipedia.org/wiki/Dependency_injection).  I have successfully integrated a [dependency injection tool](https://github.com/wessberg/DI) that uses a [transformer](https://github.com/wessberg/DI-compiler), which has a tiny footprint (excellent for microcontrollers) as it does most of it's work at compile time.  

**Describe alternatives you've considered**

I have a script that injects changes into `mcconfig.js` to have the features I require, and operate that script prior to building the tools (such as anytime I update to latest Moddable).

cmidgley, 2022-01-21:
As further evidence of this need, I now also need to add `"importsNotUsedAsValues": "preserve"` to `compilerOptions` so that my type-only imports will not be elided by TypeScript (needed to register library types for dependency injection).  

Under the proposal above, my `manifest.json` (fragment) might look like this:

```json
{
    "compiler": "npx ttsc",
    "tsconfig": {
        "types": [
            "src/esp32/typings/moddable-global",
            "src/esp32/typings/modules"
        ],
       "plugins": [
           { "transform": "./node_modules/@cmidgley/di/ttypescript/src/transformer.js" }
       ],
       "importsNotUsedAsValues": "preserve"
    }
}
```

phoddie, 2022-01-22:
I took at run at this. I think it generates what you've described. Here's a [patch](https://github.com/Moddable-OpenSource/moddable/files/7918797/patch.txt) to try. It should work with the current Moddable SDK [top-of-tree](https://github.com/Moddable-OpenSource/moddable/commit/eecb298fb1d9019854591515664e369ad9cf5b53). My experience with TypeScript is limited so it would help if you could give this a try.

All TypeScript related configuration in manifests is moved to "typescript" section in the manifest. Support for "tsconfig" at the top level is gone, so manifests need to be updated.

Here's an example based on your fragment above:

```json
"typescript": {
	"compiler": "npx ttsc",
	"tsconfig": {
		"compilerOptions": {
			"types": [
				"$(MODDABLE)/typings/base64"
			],
			"plugins": [
				{ "transform": "./node_modules/@cmidgley/di/ttypescript/src/transformer.js" }
			],
			"importsNotUsedAsValues": "preserve"
		}
	}
}
```

It is bit verbose, but allows strictly matching the tsconfig.json structure which seems likely to cause the fewer issues moving forward.

As before, paths in tsconfig.compilerOptions.types are resolved before being written to tsconfig.json. Other paths are not, such as those in plugins/transform. I'm not sure if that's a problem, but there isn't a general way for mcconfig to know which strings in a tsconfig.json fragment are paths.

The xs.d.ts typings are no longer included by mcmanifest. That's moved to manifest_base.json. You'll need to apply that change manually as it isn't in the patch.

```json
"typescript": {
	"tsconfig": {
		"compilerOptions": {
			"types": [
				"$(MODDABLE)/xs/includes/xs"
			]
		}
	}
},
```

cmidgley, 2022-01-24:
Moving discussion from Gitter....

This patch is working overall well.  One bug, when `typescript` section is missing `tsconfig` there is a error thrown:

```JSON
"typescript": {
   "compiler": "ttsc"
}
```

```
### TypeError: cannot coerce undefined to object
```

I am also having a problem getting pathing to work.  Given this source:  

```json
  "typescript": {
    "compiler": "ttsc",
    "tsconfig": {
      "compilerOptions": {
        "plugins": [
          {
            "transform": "./node_modules/@cmidgley/di/ttypescript/src/transformer.js"
          }
        ]
      }
    }
  }
  ```
  
  The path for `transform` is somewhere deep in the Moddable output directories.  If we could use a variable, similar to `$(MODDABLE)` which references  the project root (or for that matter, any environment variable) then the line would become:
  
```json
"transform": "${PROJECT_DIR}/node_modules/@cmidgley/di/ttypescript/src/transformer.js"
```

`$PROJECT_DIR` could then be set in an environment variable, or perhaps even on the `mcconfig` command line:

```bash
mcconfig -m -d -p esp32 PROJECT_DIR=`pwd`
```

The above works on all platforms, where as the following only works on Linux/Mac:

```bash
PROJECT_DIR=`pwd` mconfig ...
```

---

The above was posted in the Gitter discussion, where Peter commented:
> The "transform" path is evaluated by TypeScript, not mcconfig. I don't know if/how TypeScript applies environment variables to paths in tsconfig .

A quick google indicates `tsconfig` will not expand variables.  Does mcconfig (etc) currently do any macro expansion, or is that all done by the surrounding services (makefiles, etc)?  If it does, then can we apply that to the source string before merging into the JSON object?

phoddie, 2022-01-24:
`mcconfig` does apply environment variables to path strings. That's how $(MODDABLE) works.  From the notes above:

> ...paths in tsconfig.compilerOptions.types are resolved before being written to tsconfig.json. Other paths are not, such as those in plugins/transform. I'm not sure if that's a problem, but there isn't a general way for mcconfig to know which strings in a tsconfig.json fragment are paths.

Resolving the environment variables in path strings is easy. Knowing which strings are paths is not. The paths in "transform" are an extension to tsconfig.json, so how could `mcconfig` know to map them?

cmidgley, 2022-01-24:
Does it matter if it is a path to do environment variable expansion?  I was just thinking to expand any usage of an environment variable, much like how a makefile or shell script works.

I have a workaround in place, which takes advantage of the new `compiler` feature.  It executes a shell script instead of `tsc`, and in there pre-processes `tsconfig.json` to substitute $(PROJECT_DIR)` for the desired directory.  Similar end result (substitution with concern for context), but not as customizable / obvious.

cmidgley, 2022-01-24:
Something to be aware of ... when having these settings in multiple manifest files, the merge doesn't always result in a working `tsconfig`.  For example, the array `plugins` may end up with many `transform` entries if defined in multiple manifests:

```json
        "plugins": [
          {
            "transform": "./node_modules/@cmidgley/di/ttypescript/src/transformer.js"
          },
          {
            "transform": "./node_modules/@cmidgley/di/ttypescript/src/transformer.js"
          },
          {
            "transform": "./node_modules/@cmidgley/di/ttypescript/src/transformer.js"
          }
        ]
```

(which happens to cause `tsc` to crash!)

This is a classic JSON merge issue (arrays specifically) so unless you feel otherwise, this doesn't feel like something `mcmanifest` (etc) should try to resolve.

phoddie, 2022-01-24:
That's not surprising: merging unknown objects is even more impractical than identifying paths. There are limits to what we can do at this level.

phoddie, 2022-01-24:
> Does it matter if it is a path to do environment variable expansion? I was just thinking to expand any usage of an environment variable, much like how a makefile or shell script works.

The risk is applying an expansion where we should not. mcconfig and mcrun currently only apply expansions to paths. This conservative approach minimizes the risk of surprises.

For tsconfig.compilerOptions.plugins we could just apply environment variables to every string. It might be safe. ;) I really can't say.

cmidgley, 2022-01-24:
Given that no macro expansion currently occurs on tsconfig.json, the only risk I can see is if a syntax of `$(NAME)` needed to survive unexpanded in tsconfig.  I don't know of any such condition given that the file is mostly generated by `mcconfig`.  The risk could be further reduced if the expansion left the syntax alone if the variable doesn't exist, so only those variables defined would expand ... but that I really think is in the weeds and hyper-optimizing something that doesn't feel like a real issue.  

I did a google search of github, looking for all tsconfig.json files that have "$(".  Seven total matches, none using it as we discuss.  To ensure my search was legit, I did it again but with "bash" instead of "tsconfig.json" and hit millions.

Feels pretty safe to do it to all of `typescript` to me...?

phoddie, 2022-01-24:
> Feels pretty safe to do it to all of typescript to me...?

We can try. I'll code something up. It will eliminate the special case for tsconfig.compilerOptions.types.

FWIW – If we want / need to provide control over the substitution behavior in the future, that could be done in the `typescript` object outside of `tsconfig`.

phoddie, 2022-01-24:
This [revised patch](https://github.com/Moddable-OpenSource/moddable/files/7928833/patch.txt) substitutes environment variables into all strings in the tsconfig element using a JSON roundtrip with a reviver. I'm not proud of that. ;) Maybe it will give the behavior you want?

cmidgley, 2022-01-24:
Patched applied and it all works great now.  Really appreciate all your time on this.

phoddie, 2022-01-24:
Phew.

phoddie, 2022-01-26:
This [commit](https://github.com/Moddable-OpenSource/moddable/commit/f6acf1b74a7b4dc10d1e89121af7f91d5087980d) integrates the patch.

cmidgley, 2022-01-26:
Integrated and confirmed.  Thank you.

closed 2022-01-26 by cmidgley