Moddable-OpenSource/moddable #773: Unable to execute mcrun

js4iot, 2022-01-10:
**Build environment:** Windows on a VirtualBox virtual machine 
**Target device:** I have a cheap [generic Wroom ESP32 board](https://www.amazon.ca/gp/product/B08YJCF6GF/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1), I am targeting "esp32".

**Description**
I am able to build the Hello World sample and this works great, but I am not able to make a small modification without rebuilding the entire project.

**Steps to Reproduce**
1. Build and install the sample Hello World app using this build command: `mcconfig -d -m -p esp32`. This works great!
2. Make a small modification to the code
3. Execute `mcrun -d -m -p esp32`

**Expected behavior**
I expect this to rebuild the JavaScript code, taking much shorter than `mcconfig`

**Other information**
Instead, this is what I get:

```
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:532
load:0x40078000,len:13508
load:0x40080400,len:3500
entry 0x400805d0
Installing mod.### fxCommandReceived: remote operation failed with resultCode 65535
NMAKE : fatal error U1077: 'C:\js4IoT\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0xffffffff'
Stop.
```

cmidgley, 2022-01-10:
It sounds like you are trying to build the base firmware with `mcrun`.  That is not how `mcrun` works - it is used to build mods that are loaded by your base firmware.  The base firmware, sometimes called the Mod Host, has all the compiled in libraries (anything that depends on C) plus whatever JS code you want.  Your manifest in that code needs to specify that it wants to dynamically load mods (defines `XS_MODS`).  Then you can have your program, in the base firmware that is compiled by `mcconfig` and flashed into the device, check if a mod is available and execute any modules within that mod that you want.  Those modules are the ones that are compiled and delivered with `mcrun`.

See the [mods documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/mods.md) for additional information (and the book, as it does a much better job with describing this process).

phoddie, 2022-01-10:
@cmidgley is correct. `mcrun` and `mcconfig` are not interchangeable.

js4iot, 2022-01-11:
Just to make sure we are on the same page, the code that I  modified was the hello world sample. I made a small change to the JavaScript code. I changed the message displayed in the trace() function.

I want to send this new code (a small change to the JavaScript code) to the esp32 board very quickly without having to do a full build with mcconfig which takes a long time. Is that possible? From what I have read, I thought me fun was supposed to do that.

phoddie, 2022-01-11:
> Is that possible?

Yes, but not with the `$MODDABLE/examples/helloworld` project.

@cmidgley provided a good explanation and suggested two good resources for further reading.

cmidgley, 2022-01-11:
It is possible ... but not the way you are doing it.  Basically the ESP32 needs a host firmware that has the JVM and whatever compiled JavaScript code you want.  That code, which in your case includes your 'hello, world' application, can only be compiled using `mcconfig` and then fully flashed.  

However, you can create that base application (along with whatever library code you want included in the binary, which you specify in your `manifest.json`) and enable "mods".  Your JS code then can request to load a module (import it) that is contained in a mod, and that code can be installed with `mcrun` and is very fast.  

Think of it this way ... the base image is like an operating system.  You can load programs on top of it without reinstalling the OS .  But if you need to change the OS, you need to rebuild it.  In the case of Moddable, you build the OS as your main program and flash it - including some amount of your JavaScript code.  Then you can install one program on top of this -- called a "mod".  That mod can have any number of JavaScript modules compiled with it, and it's pure byte-code.  Your OS (your JavaScript code) is responsible for deciding when and how to load and run that mod.  

The biggest restriction of the mods (vs the base firmware) is that mods can't have anything that talks directly to C code - it's bytecode only.  Which makes sense when you think about it - mods are pure JVM bytecode (and portable across processors).  The flash base image (OS in my analogy)  is native compiled code for the specific processor and therefore talk to C and low level services.

Check out their book, and the samples github page for the book.  It's totally built around this idea and makes it much more understandable.

By the way - those mods can be run as Workers (which run with RTOS preemptive multitasking, and on an ESP32 can span cores), and even with Secure ECMAScript Containers - providing for a very secure container for untrusted (mod) code to run, protecting the host.

js4iot, 2022-01-11:
Thank you!

I think that was the part I was missing in my understanding.

js4iot, 2022-01-11:
I have been reading what you sent me @cmidgley, and I think this is exactly what I need but I do not think it's working.

So I cloned [the book's samples repo](https://github.com/Moddable-OpenSource/iot-product-dev-book). And I perform these tasks:

```
cd C:\js4IoT\iot-product-dev-book\ch1-gettingstarted\host
mcconfig -d -m -p esp32 
```

This seems to work fine. It builds the host and it flashes it. I do see in xsbug this expected output `Device flashed. Ready to install apps.`. but it shows this message in the command prompt window, not sure if this is an issue that I should worry about:

```
Wrote 659888 bytes (381638 compressed) at 0x00010000 in 39.7 seconds (effective 132.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
        1 file(s) copied.
The system cannot find the file specified.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:532
load:0x40078000,len:13508
load:0x40080400,len:3500
entry 0x400805d0
```

I believe the error `The system cannot find the file specified.` is because I have not build the mod so I ignore it and I move to the next step:

```
cd  C:\js4IoT\iot-product-dev-book\ch1-gettingstarted\helloworld
mcrun -d -m -p esp32
```

I immediately get this error

```
( 9:22:46.74) C:\js4IoT\iot-product-dev-book\ch1-gettingstarted\helloworld>mcrun -d -m -p esp32
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:532
load:0x40078000,len:13508
load:0x40080400,len:3500
entry 0x400805d0
Installing mod.### fxCommandReceived: remote operation failed with resultCode 65535
NMAKE : fatal error U1077: 'C:\js4IoT\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0xffffffff'
Stop.
```

What am I doing wrong?

cmidgley, 2022-01-11:
I'll likely need to leave it to @phoddie and team to help on this, but a few comments:

1) You do not need to build a mod in advance of a base firmware.  The base firmware actually doesn't know if there will ever be a mod.  
2) Getting the message that the base firmware is `Ready to install apps` means you got that base firmware loaded and ready, so likely that `File not found` message is benign (though I'll have to see if I get it).
3) By chance have you recently updated Moddable without recompiling from scratch?  Sometimes errors on mod loading occur due to incompatible byte-code versions between base and the mod (though I've seen those at runtime in the firmware, not externally from mcrun)
4) Did you modify the sample files at all?  If yes, perhaps try them 100% unmodified first.

I'm doubtful any of the above will help you solve this problem though.  I'm going to be doing some more mod tests hopefully later today, so I'll pay closer attention to the output and if anything jumps out I'll let you know.

js4iot, 2022-01-11:
@cmidgley, thanks for getting back to me.

1. I did not build the mod before the host.
2. I also agree it should be benign, just a warning not an error. And because I just flashed the hows and I have not added the mod then it would make sense.
3. I have not made any changes to the Moddable code, or the book sample code. Actually, this is a new virtual machine, which was created to just have a clean environment with nothing else added or removed.
4. I have not made any changes to the Moddable code, or the book sample code

js4iot, 2022-01-11:
Actually, on number 3... I did **not** follow [these instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#windows) to install ESP-IDF.  I found a much simpler way to install it: https://dl.espressif.com/dl/esp-idf/?idf=4.4

phoddie, 2022-01-11:
>  believe the error The system cannot find the file specified. is because I have not build the mod...

It is not. As @cmidgley notes, the host has no dependency on or knowledge of the mod. 

> Installing mod.### fxCommandReceived: remote operation failed with resultCode 65535

This error can indicate an attempt to install a mod into a host that doesn't support mods. For example, if $MODDABLE/examples/helloworld was installed when executing `mcrun`. 

FWIW – I just confirmed that I am able to run the Chapter 1 [host](https://github.com/Moddable-OpenSource/iot-product-dev-book/tree/master/ch1-gettingstarted/host) and its [helloworld mod](https://github.com/Moddable-OpenSource/iot-product-dev-book/tree/master/ch1-gettingstarted/helloworld) with current sources on an ESP32 and in the simulator on macOS.

js4iot, 2022-01-11:
> This error can indicate an attempt to install a mod into a host that doesn't support mods.

How would I know if my [generic ESP32 board](https://www.amazon.ca/gp/product/B08YJCF6GF/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1) supports mods? As mentioned before I am not using any of the boards [listed here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#platforms).

js4iot, 2022-01-13:
I tried something today which worked, but I want you to help me understand why or if there is a flaw with my logic.

[This page](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#esp32) which lists the supported hardware there is an entry for an `esp32/nodemcu` platform. When I bought my ESP32 chip, I did not know about "NodeMCU" and that you could buy either [ESP32](https://www.amazon.ca/gp/product/B08YJCF6GF/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1) or [ESP32 NodeMCU](https://www.amazon.ca/dp/B09DP272V7/?coliid=I1N9ZMPY9LWW9T&colid=24C5MCARPTYBP&psc=1&ref_=lv_ov_lig_dp_it) boards.

But reading the [NodeMCU Documentation](https://nodemcu.readthedocs.io/en/dev-esp32/), I realized that NodeMCU is "NodeMCU is an open source Lua based firmware for the ESP32 and ESP8266" so it's not a hardware thing, but a software thing. So any ESP32 board can be converted into an ESP32 NodeMCU board if the right software is installed.

- First question: Is this assumption correct?

I then found a [tool that flashes the NodeMCU firmware](https://github.com/marcelstoer/nodemcu-pyflasher) into a regular ESP32 board converting it into a ESP32 NodeMCU board. You must create the firmware using [this website](https://nodemcu-build.com/) which creates the firmware that includes 7 modules by default: file, gpio, net, node, tmr, uart, and wifi. But there is a big list of modules that you can include:

[image]

At this moment I do not know if I need to select any other modules, because I do not know how Moddable SDK would need those modules or if they get flashed when I build Moddable SDK, which leads me to my second question.

- Second question: How does Moddable SDK flashing mechanism uses the modules in the NodeMCU? Are there Moddable SDK versions of those modules that override the ones in the firmware?

Based on my understanding, I am visualizing this architecture.

[image]

- Third question:  Is that even remotely close to the way it works, or am I totally wrong?

Thank you so much for helping me understand this new world :-)

js4iot, 2022-01-13:
BTW, I tried it and it work!!!! I was able to get the JavaScript mod to flash and execute!!!!

cmidgley, 2022-01-13:
It's great you got it to work!  However - your understanding is incorrect of the software stack is incorrect.  While NodeMCU is software, it is also a hardware spec.  There is no Lua in the software stack with Moddable.  Flashing moddable replaces everything with custom firmware/settings to Moddable - which makes your discovery surprising to me!  

Perhaps @phoddie can help explain what it might have flashed that the default Moddable flashing didn't?  With a bit of effort you can dig into this - look at the flashing parameters from your LUA flash (`idf.py` or `esptool` is typically used for flashing), and specifically at the file/address/lengths of whatever was flashed.  Also look at the partition table that was flashed, to understand mapping of addresses to partitions as they may use a different partition scheme.  Then do the same with the Moddable flashing (build with `mcconfig -v` to see the flash command), and compare the two settings. 

More technically, Moddable also builds using ESP-IDF (like NodeMCU Lua does), but does not include any NodeMCU Lua libraries/code.  It includes flashing a bootloader, partition table, core firmware (including FreeRTOS), etc, all specific to or provided by the Moddable setup.

I've done this on tons of chips now (I'd guess around 15 models, all but two being cheap knockoffs), with never having any issues as your describe.  Many of my boards are very similar to yours (with the WROOM specification, which yours has).   Hoping someone can explain this behavior as it seems quite surprising to me!

js4iot, 2022-01-13:
Thanks for your complete answer @cmidgley.

There is one difference in the way I used `mcconfig` and `mcrun`, I used `-p esp32/nodemcu` rather than `-p esp32`. Maybe I could have done `-p esp32/nodemcu` all along but my board is not a "NodeMCU" so I went with the default `esp32`.

I would love to test this and see if there are any differences, and I will report my findings. In order to test this, I want to clear the board and do erase **everything** from my board. Does`idf.py --verbose --baud 115200 --port COM3 erase_flash` clear everything?

Do you have some information as to what are the hardware specifications of a NodeMCU board and how they are different from an ESP32 board? I have found this, but thing about hardware.

> NodeMCU is an open source Lua based firmware for the ESP32 and ESP8266 WiFi SOC from Espressif and uses an on-module flash-based SPIFFS file system. NodeMCU is implemented in C and is layered on the Espressif ESP-IDF.
https://nodemcu.readthedocs.io/en/dev-esp32/

Good to know that you have used multiple boards successfully with Moddable SDK. I have only used one and I have had nothing but problems :-)

In Moddable defence, I am new to MCU, I have done some Raspberry Pi before and most of my experience has been as a developer, recently (last 5 years) JavaScript developer (Fron-end and Node).

cmidgley, 2022-01-13:
Technically you shouldn't need to "erase the entire board".  Flashing with bootloader, partition table and partition content does that (with a possible exception, see below).  I've never needed to erase the entire board, and so I'm unsure what the technically correct syntax is to pull that off fully.  

When I'm just using ESP32 generically, I use use `mcconfig -m -d -p esp32`.  This works on all my boards.  For some, such as the boards that have LoRa radios, I need to specify a more specific platform so additional driver setup is available ... though even that I can ignore if I just do the setup in my code directly.  

My understanding of an ESP32 vs. NodeMCU really has to do with the hardware side of the world - pinout of gpios, power/gnd, USB -> Serial, etc.  There are differences in the ESP32 chip - how many cores, which generation chip it is.  But I would have thought almost any ESP32 board would work (for a supported chip family) assuming you can get the serial port connected.  One thing to know is that RAM and flash can be different between boards - the defaults in Moddable work on most everything but your board may have more room.  Those parameters can be configured if desired.

So back to your problem / discovery ... Have you gone back to try it with `mcconfig -m -d -p esp32` now?  My bet is that it works fine.  If I recall the details correctly, you had the board working fine ... just without mods.  That really tells us everything was fine with the board.  Then you tried to do mods and ran into problems where they didn't load (-1 error) and Peter said some boards are not supported.... and down the drain we went, right?

I'm now thinking flashing NodeMCU didn't really flash anything interesting that needed to be there.  I'm guessing instead the mod partition had some value that caused Moddable to get into trouble, and loading NodeMCU overwrote the mod area (likely even a different partition table) making it work the next time you flashed it.  Peter may disagree ... or not?  I've never asked him about it.

The reason I'm thinking that is I had a similar experience a while ago (only once, but for days on only one device)... and was the one time I did do a partial erase.  I couldn't get a mod to load and run.  The host worked if I didn't enable mods, but if I did enable them it was ... unreliable?  And just wouldn't load my mods.  I eventually force erased the mod partition, and suddenly I was back up and working again.  I was never able to reproduce it, so I never submitted any bug report.  After many hundreds (thousands?) of flashings, it's only happened that one time, on one device (in my case, a well used/loved device, not a brand new one).