Moddable-OpenSource/moddable #774: SEGV xs/sources/xsDataView.c:559:24 in fx_ArrayBuffer_prototype_concat

hope-fly, 2022-01-10:
##### Moddable-XS revision

Commit: [2f93df29](https://github.com/Moddable-OpenSource/moddable/commit/2f93df297782cf0a31c1797b246becbf5e546f4f)

Version: 11.5.0 32 4

##### Build environment

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

##### Build steps

```bash
cd  ~/moddable/xs/makefiles/lin
make -f xst.mk
```

##### Test case

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
function JSEtest(f, args) {
    var bound = Function.prototype.bind.apply(f, new ArrayBuffer(64).concat(new Object));
    return new bound();
}
var d = JSEtest(Date, [1957, 4, 27]);
```
</code></pre>
</details>

##### Execution  & Output

```bash
$ ./moddable/build/bin/lin/debug/xst poc.js

AddressSanitizer:DEADLYSIGNAL
=================================================================
==62908==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x00000061df67 bp 0x7fff84c54690 sp 0x7fff84c544f0 T0)
==62908==The signal is caused by a READ memory access.
==62908==Hint: address points to the zero page.
    #0 0x61df66 in fx_ArrayBuffer_prototype_concat /root/moddable/xs/sources/xsDataView.c:559:24
    #1 0x84f3ca in fxRunID /root/moddable/xs/sources/xsRun.c:842:7
    #2 0x8ceaac in fxRunScript /root/moddable/xs/sources/xsRun.c:4766:4
    #3 0xad3231 in fxRunProgramFile /root/moddable/xs/tools/xst.c:1387:2
    #4 0xacfa83 in main /root/moddable/xs/tools/xst.c:281:8
    #5 0x7f1a84524bf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #6 0x42ddc9 in _start (/usr/local/bin/xst+0x42ddc9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/moddable/xs/sources/xsDataView.c:559:24 in fx_ArrayBuffer_prototype_concat
==62908==ABORTING
```

phoddie, 2022-01-10:
Thank you for the report.

This PoC can be simplified:

```js
new ArrayBuffer(64).concat({})
```

FWIW – `Array.prototype.concat` is an extension to JavaScript by XS, not part of EcmaScript standard.

phoddie, 2022-01-10:
It looks like the problem is missing braces. 🤦

https://github.com/Moddable-OpenSource/moddable/blob/83dadd3def6d2e7e75fc003a5ab409aa81275dd8/xs/sources/xsDataView.c#L557-L559

hope-fly, 2022-01-10:
that's interesting, I'll note it 😉

phoddie, 2022-01-10:
It looks like that problem snuck in with recent changes to the internal XS buffer representation. I imagine some linter would pick that mistake up.

hope-fly, 2022-01-13:
Verified!

closed 2022-01-13 by phoddie