Moddable-OpenSource/moddable #775: Manifest incorrectly generates tsconfig.json when targeting a .d.ts file

cmidgley, 2022-01-10:
**Build environment:** Windows and Linux
**Target device:** ESP32

**Description**

When using a manifest that targets a named  `.d.ts` file, the module name in `tsconfig.json` has two characters stripped from it.  For example, from a library manifest file such as this:

```json
{
	"modules": {
		"@scope/mylib":  [ "./demo", "./demo.d" ]
	}
}
```

the resulting generated `tsconfig.json` under `paths` looks like this:

```json
"paths": {
	"@scope/myl": [
		"/root/Projects/moddable/examples/helloworld/mylib/demo"
	],
	"main": [
		"/root/Projects/moddable/examples/helloworld/main"
	]
},
```

Note that the package name should be `@scope/mylib` not `@scope/myl`.  I have tried this with and without using a scope, and the problems remains the same.

**Steps to Reproduce**

1. Clone repo https://github.com/cmidgley/moddable-manifest-typescript-issue
2. Compile (it will fail on typescript)
3. Inspect generated `tsconfig.json`

phoddie, 2022-01-26:
This module fragment isn't correct:

```json
"modules": {
	"@scope/mylib": ["./demo", "./demo.d"]
}
```

There are a couple of options, depending on your goals and preferences.

Wildcards allow for a concise manifest:

```json
	"modules": {
		"@scope/mylib*": "./demo*"
	}
```

For this project, that is equivalent to the following, which also works.

```json
{
	"modules": {
		"@scope/mylib": "./demo",
		"@scope/mylib.d": "./demo.d"
	}
}
```

cmidgley, 2022-01-26:
That works!  I have removed my `sed` hack, and successfully built and tested the 8 Moddable builds I currently have with it.

Once again - thank you so much for the amazing help / support.  

---

Side note:  This doesn't affect me, but worth noting in case it comes up in the future.   If you specify only a `.d.ts` file, such as:

```json
"modules": {
   "@scope/test": "./mytest.d"
}
```

The resulting `tsconfig.json` is mangled:

```json
"paths": {
   "@scope/te": "<path>"
}
```

closed 2022-01-26 by cmidgley

phoddie, 2022-01-26:
Glad this works for you.

On the `.d` I think you need to write that as shown above:

```
	"modules": {
		"@scope/test.d": "./demo.d"
	}
```