Moddable-OpenSource/moddable #776: XSC crashes with segmentation fault (Linux) or fatal error U1077 (Windows)

cmidgley, 2022-01-11:
**Build environment:** Confirmed on Linux and Windows
**Target device:** ESP32

**Description**
When compiling a specific module that has pre-compiled typescript (and therefore it has `js` files with maps), `xsc` crashes with a `Segmentation fault` or `U1077`:

Linux:
```
# xsc @cmidgley/di-container/di-container.xsb
make: *** [/workspace/apps/firmware/compiled/tmp/esp32/heltec_lora_32/debug/firmware/makefile:869: /workspace/apps/firmware/compiled/tmp/esp32/heltec_lora_32/debug/firmware/modules/@cmidgley/di-container/di-container.xsb] Segmentation fault
The script failed with exit code 2
```

Windows:
```
# xsc @cmidgley\di-container\di-container.xsb
NMAKE : fatal error U1077: 'c:\moddable\build\bin\win\release\xsc.EXE' : return code '0xc0000005'
Stop.
```

**Steps to Reproduce**

1. Clone https://github.com/cmidgley/moddable-xsc-segfault
2. Compile with `mcconfig -m -d -p esp32`

**Additional Information**

If the file `di-container.js` has it's ``//# sourceMappingURL=di-container.js.map`` removed, or if the source file referenced by the map (`di-container.ts`) is renamed/deleted, it will compile without segmentation fault.

If you do get the files to compile, you will likely get a TypeScript error (see #775).  Ignore that error, as if you get that far you have gotten past the segmentation fault issue.

phoddie, 2022-01-11:
Congratulations on finding a way to reproduce it. Unfortunately, it didn't happen on macOS for me. I enabled ASAN for xsc and it did point out a problem. I'm still not sure about the root cause or a correct fix, but a quick hack resolved the crash. In xsSourceMap.c try changing line 115 to:

```
parser->lines = fxNewParserChunkClear(parser, (1 + line) * sizeof(txInteger));
```

Then rebuild xsc. Does that change anything for you?

cmidgley, 2022-01-11:
Thanks for looking at this so quickly!  I just checked, but that line is the same as the current line 115...?

phoddie, 2022-01-11:
Oops..... the changed version is:

```
parser->lines = fxNewParserChunkClear(parser, (100 + line) * sizeof(txInteger));
```

cmidgley, 2022-01-12:
Verified in both my test project and in my main project, including execution on the ESP32.  Thank you so much!  I'd appreciate it if you could let me know (here or gitter.im) when this and the other xsc patch are available in the github, and I'll rebuild then.

Thanks!

phoddie, 2022-01-17:
The root cause of this issue, tracked down by @patrick-soquet, is that some line numbers were mapped through the source map twice. There's a fix pending for that. It seems likely that the change will also improve the debugging of source files with source maps since there have been incorrect mappings.

cmidgley, 2022-01-17:
Great - I'll give it a whirl when it's pushed.  Any chance this could help with #753?  Guessing not as this is an `xsc` issue and it feels like #753 is an `xsbug` issue, but forever hopeful...!

phoddie, 2022-01-17:
#753 seems unrelated, but I would be happy to be proven wrong.

phoddie, 2022-01-18:
The fix for this is now available.

closed 2022-01-18 by cmidgley