Moddable-OpenSource/moddable #792: SEGV xsAPI.c:948 in fxGetAll

jessysaurusrex, 2022-01-24:
### Version: 
[eecb298fb1d9019854591515664e369ad9cf5b53](https://github.com/Moddable-OpenSource/moddable/tree/eecb298fb1d9019854591515664e369ad9cf5b53)

### poc.js
```js
function main() {
const v0 = -1589470555n;
const v1 = "object";
const v3 = "function";
const v4 = 9419;
const v6 = new Compartment();
const v7 = v6.module();
const v8 = 0;
const v9 = 2;
const v10 = 1;
function* v11(v12,v13) {
}
const v16 = new Int32Array(36251);
try {
    const v17 = v16.reduce(v11);
} catch(v18) {
}
const v19 = [-2432330610];
const v20 = {};
const v21 = [v20,v20];
const v23 = [690473.6494891534,690473.6494891534];
const v24 = Set;
const v26 = Date();
function v27(v28,v29) {
    'use strict';
    const v31 = [];
    const v32 = Reflect.apply();
    const v33 = [v31];
    const v34 = "boolean";
    const v35 = Int8Array;
    const v36 = Array;
    const v37 = v33.delete;
    const v38 = Reflect.isSafeInteger(v37,v23);
}
const v39 = Promise;
const v42 = /4\Dp+/gis;
for (let v43 = v23; v43 >= -536870912; v43 = v43 + 2157069075) {
}
}
main();
```

### Invocation `xst poc.js`
```
STDERR: 
/Users/amulet/moddable/xs/sources/xsScript.c:203:16: runtime error: left shift of 1327365517 by 1 places cannot be represented in type 'txSize' (aka 'int')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsScript.c:203:16 in 
/Users/amulet/moddable/xs/sources/xsLexical.c:333:21: runtime error: 2.43233e+09 is outside the range of representable values of type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsLexical.c:333:21 in 
/Users/amulet/moddable/xs/sources/xsDataView.c:1439:42: runtime error: left shift of 3 by 30 places cannot be represented in type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsDataView.c:1439:42 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==90811==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000301 (pc 0x000100c0139c bp 0x0001012ced90 sp 0x00016f1e4790 T0)
==90811==The signal is caused by a UNKNOWN memory access.
==90811==Hint: address points to the zero page.
    #0 0x100c0139c in fxGetAll xsAPI.c:948

==90811==Register values:
 x[0] = 0x00000001085a45a0   x[1] = 0x000000000000000b   x[2] = 0x0000000000000000   x[3] = 0x000000010830ad60  
 x[4] = 0x000000702de56a00   x[5] = 0x0000000000000001   x[6] = 0x0000000000000000   x[7] = 0x0000000000000001  
 x[8] = 0x0000000000000301   x[9] = 0x0000000000000301  x[10] = 0x0000000000000000  x[11] = 0x0000000000000133  
x[12] = 0x000000016f1ff228  x[13] = 0x000000016f207da0  x[14] = 0x00000000f7b9517b  x[15] = 0x00000000f7b9517b  
x[16] = 0x00000001c41ec110  x[17] = 0x0000000101f705b8  x[18] = 0x0000000000000000  x[19] = 0x0000000101964060  
x[20] = 0x0000000101466a58  x[21] = 0x0000000101b98070  x[22] = 0x0000000000000000  x[23] = 0x0000000000000000  
x[24] = 0x0000000000000000  x[25] = 0x0000000000000000  x[26] = 0x0000000000000000  x[27] = 0x0000000000000000  
x[28] = 0x0000000000000000     fp = 0x000000016f1e4ce0     lr = 0x00000001012ced90     sp = 0x000000016f1e4790  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV xsAPI.c:948 in fxGetAll
==90811==ABORTING

Status: 
pid 90811 SIGABRT (signal 6)
```

phoddie, 2022-01-25:
The essential part of the PoC is:

```js
function* v11(v12,v13) {
}
const v16 = new Int32Array(36251);
try {
    const v17 = v16.reduce(v11);
} catch(v18) {
}
```

This triggers a JavaScript stack overflow when the reducer function is a generator. XS detects the overflow and throws a `RangeError`.  Because the error is caught, execution continues and eventually fails

When xst is used to run test262,, the exception would be unhandled which would cause the test to exit cleanly and report an error. For fuzzing, we may want to completely halt the test case on the JavaScript stack overflow. FWIW _ on production deployments like xsnap and the Moddable SDK runtime, execution is terminated when a JavaScript stack overflow is detected.

jessysaurusrex, 2022-02-02:
Verified fixed in `fb7b2c2`

closed 2022-02-02 by jessysaurusrex