Moddable-OpenSource/moddable #798: stack-overflow xsMemory.c:945 in fxMarkValue

jessysaurusrex, 2022-01-24:
### Version: 
[eecb298fb1d9019854591515664e369ad9cf5b53](https://github.com/Moddable-OpenSource/moddable/tree/eecb298fb1d9019854591515664e369ad9cf5b53)

### poc.js
```js
function main() {
async function* v1(v2,v3) {
    return resourceName;
}
const v7 = Date();
function v8(v9,v10) {
    const v12 = [];
    const v13 = Reflect.apply(v9,v7,v12);
    function v14(v15,v16) {
        const v18 = `object${v13}toString${Date}number${v16}65536${Number}9007199254740992`;
        with (Object) {
            toString = v9;
        }
    }
    const v19 = v8(v14);
}
const v21 = new Promise(v8);
}
main();
```

### Invocation `xst poc.js`
```
STDERR: 
/Users/amulet/moddable/xs/sources/xsScript.c:203:16: runtime error: left shift of 1327365517 by 1 places cannot be represented in type 'txSize' (aka 'int')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsScript.c:203:16 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==90827==ERROR: AddressSanitizer: stack-overflow on address 0x00016af47ff8 (pc 0x0001049debd8 bp 0x0001049f0904 sp 0x00016af472c0 T0)
    #0 0x1049debd8 in fxMarkValue xsMemory.c:945

SUMMARY: AddressSanitizer: stack-overflow xsMemory.c:945 in fxMarkValue
==90827==ABORTING

Status: 
pid 90827 SIGABRT (signal 6)
```

phoddie, 2022-01-25:
Sloppy mode only.

phoddie, 2022-01-25:
Stack overflow because of recursive marking of environment. Sloppy only because of `with`.

jessysaurusrex, 2022-02-02:
Verified fixed in `fb7b2c2`

closed 2022-02-02 by jessysaurusrex